WPIRoboticsProjects/GRIP #365: Resize operation issues

gruffen, 2016-01-15:
cv_resize throws an error, not sure if that's why it's not possible to resize with negative or fractional values in fy/fx; preview is not visible

Message: The CV Resize operation did not perform correctly.

Image of my pipeline: 

[image]
The resulting preview: 
[image]

## System Info

Property Name | Property
---------------------|----------------
GRIP version | 1.0.0
OS version | Windows 10
arch | amd64
javafx version | 8.0.60
java version | 1.8.0_60

## Stack Trace 

    java.lang.RuntimeException: src\imgwarp.cpp:3209: error: (-215) dsize.area() > 0 || (inv_scale_x >   0    && inv_scale_y > 0) in function cv::resize

        at org.bytedeco.javacpp.opencv_imgproc.resize(Native Method)
	at edu.wpi.grip.generated.opencv_imgproc.Resize.perform(Resize.java:58)
	at edu.wpi.grip.core.Operation.perform(Operation.java:61)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:137)
	at edu.wpi.grip.core.Step.onInputSocketChanged(Step.java:156)
	at sun.reflect.GeneratedMethodAccessor31.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.Socket.setValueOptional(Socket.java:70)
	at edu.wpi.grip.core.Socket.setValue(Socket.java:79)
	at edu.wpi.grip.ui.pipeline.input.NumberSpinnerInputSocketController.lambda$new$77(NumberSpinnerInputSocketController.java:52)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:349)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.ObjectPropertyBase.fireValueChangedEvent(ObjectPropertyBase.java:105)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(ObjectPropertyBase.java:112)
	at javafx.beans.property.ObjectPropertyBase.set(ObjectPropertyBase.java:146)
	at javafx.scene.control.SpinnerValueFactory.setValue(SpinnerValueFactory.java:150)
	at edu.wpi.grip.ui.util.Spinners.lambda$makeEditableSafely$118(Spinners.java:56)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:361)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.ReadOnlyBooleanPropertyBase.fireValueChangedEvent(ReadOnlyBooleanPropertyBase.java:72)
	at javafx.scene.Node$FocusedProperty.notifyListeners(Node.java:7718)
	at javafx.scene.Node.setFocused(Node.java:7771)
	at com.sun.javafx.scene.control.skin.ComboBoxPopupControl$FakeFocusTextField.setFakeFocus(ComboBoxPopupControl.java:526)
	at com.sun.javafx.scene.control.skin.SpinnerSkin.lambda$new$472(SpinnerSkin.java:135)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:361)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.ReadOnlyBooleanPropertyBase.fireValueChangedEvent(ReadOnlyBooleanPropertyBase.java:72)
	at javafx.scene.Node$FocusedProperty.notifyListeners(Node.java:7718)
	at javafx.scene.Scene$13.invalidated(Scene.java:2077)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(ObjectPropertyBase.java:111)
	at javafx.beans.property.ObjectPropertyBase.set(ObjectPropertyBase.java:146)
	at javafx.scene.Scene$KeyHandler.setFocusOwner(Scene.java:3924)
	at javafx.scene.Scene$KeyHandler.requestFocus(Scene.java:3971)
	at javafx.scene.Scene$KeyHandler.access$1900(Scene.java:3910)
	at javafx.scene.Scene.requestFocus(Scene.java:2044)
	at javafx.scene.Node.requestFocus(Node.java:7879)
	at com.sun.javafx.scene.control.behavior.ButtonBehavior.mousePressed(ButtonBehavior.java:156)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:95)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:89)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$355(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$149(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

JLLeitschuh, 2016-01-15:
Try decimal values. I usually use `.3` and `.3` for the Mac iSight camera.

gruffen, 2016-01-15:
I'm unable to input fractional values, only whole numbers. would that be exclusive to camera feed images?

JLLeitschuh, 2016-01-15:
Sorry, you are going to have to enter a zero first. So actually the numbers would be `0.3` and `0.3`
Related to #329

closed 2016-01-21 by ThomasJClark