WPIRoboticsProjects/GRIP #367: When trying to deploy GRIP to roboRio it fails everytime.

markebert, 2016-01-15:
Whenever we try to deploy GRIP to our roboRio it always fails. We have tried deploying wirelessly and have also tried do to it through the mDNS address and the IP address and have also tried it while plugged directly into the roboRio with the USB type B cable and its IP address and we still cannot successfully deploy to the roboRio. Running the program from the laptop however works fine as it is able to successfully publish the contours report to the network table but we have tried seemingly everything when it comes to attempting to deploy to the roboRio so that we can have it run in headless mode.

ThomasJClark, 2016-01-15:
Mind posting whatever error you get?

markebert, 2016-01-15:
In the deploy window it would say something like "Failed 0/5" and then would latter end at like "Failed 31/5" and then say Failed to connect. We have also tried checking the roboRio's file system to see if the files like grip.jar were successfully pushed into it but the folders were left unaltered. I wish I could give you a better example of the error but I dont have the things with me to give you exactly what it says.

JLLeitschuh, 2016-01-15:
You need to be connected to the robot via the ethernet cable.
Also, what OS are you running?

markebert, 2016-01-15:
I am running Windows 10 Pro x64. Also will deploying through the USB connected directly to the roboRio not work?

JLLeitschuh, 2016-01-15:
No, we use `scp` network protocol to deploy to the robot. USB is not supported at the moment.
I don't really know if there is an easy way to transfer files to the robot over USB from Java.

markebert, 2016-01-15:
Ok that may explain why we were unable to deploy. Also would it not be possible to deploy to the roboRio through being connected to the roboRio with the bridge set to Access Point mode and be connected to it over Wi-Fi?

ThomasJClark, 2016-01-15:
> No, we use scp network protocol to deploy to the robot. USB is not supported at the moment.

Have you tested this?  I don't see why this would prevent us from using USB

JLLeitschuh, 2016-01-15:
You can use SCP over USB?
I had no idea.
I created this wiki page about deploying. (Needs work but the basics are there)

https://github.com/WPIRoboticsProjects/GRIP/wiki/Deploying-to-Robot

markebert, 2016-01-15:
I will try deploying GRIP to our roboRio again tomorrow over ethernet and will let you know if I have anymore issues.

markebert, 2016-01-16:
After attempting to deploy GRIP to the roboRio again today it still failed even when done from a new installation of GRIP on a completely different PC. It did however start to deploy once but the program immediately encountered an error and then failed. I do not have the the error log from it however at the moment. I will post a more detailed response of what happened this Monday though.

ThomasJClark, 2016-01-17:
FYI: since there are several open issues and CD posts about problems deploying, I've opened [a new issue](377) where we can collect all of the information in one place.  If anyone has new information, please post it there.

closed 2016-01-17 by ThomasJClark

ThomasJClark, 2016-01-22:
[v1.1.0](https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.1.0) is out - this should fix most deploying problems