WPIRoboticsProjects/GRIP #370: Windows Server 2008 R2 Support

FlamingSpork, 2016-01-15:
I'm having an issue running GRIP 1.0.0 on my Windows machine.
I am running Java JDK 8u66 (x64) on Windows Server 2008 R2 (which is practically Win7).

When I launch GRIP, I get a message that says `Error invoking method.`
After hitting OK, I get a message that says `Failed to launch JVM`

PaulaRudy, 2016-01-15:
I'm sorry to hear you're having trouble running GRIP. Unfortunately, we don't currently support Windows Server 2008 R2, but we'll do our best to help. Is there a GRIP.log file in your home directory?

FlamingSpork, 2016-01-16:
[GRIP.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/92714/GRIP.log.txt)
Here it is.
Windows Server 2008 R2 is basically Windows 7, so it's a bit odd that it doesn't work.

JLLeitschuh, 2016-01-16:
@ThomasJClark and I were looking at your log file and really have no idea why that's happening.
It seems that you are having a problem loading the dependency injection framework that we really don't have control over.
We will tinker with it a bit. I have no idea how to reproduce this problem though.

FlamingSpork, 2016-01-16:
Okay.  I'm also getting a similar issue with a similar log output under Ubuntu 12.04 (x64) and Java JRE  1.8.0_66 on the same machine.
[GRIP.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/93111/GRIP.log.txt)

FlamingSpork, 2016-01-19:
I'm mildly worried that it's a hardware fault, as it fails to work under both Windows and Ubuntu (14.04) on the same machine.

```AMD Athlon x4 860K (3.7 GHz) and 4 GB of RAM```
With these specs, it's doubtful that I don't have sufficient processing power/memory to run it.

FlamingSpork, 2016-01-23:
I updated to GRIP 1.1.1 on the Ubuntu 14.04 x64 install on that machine, but it doesn't launch.
```
/opt/GRIP$ ./GRIP
Jan 23, 2016 5:49:46 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 23, 2016 5:49:46 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
platform: /Linux/amd64/
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM
```
[GRIP.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/102072/GRIP.log.txt)

ThomasJClark, 2016-01-23:
Is there any chance you could try on Ubuntu 15.10?

FlamingSpork, 2016-01-23:
I only just recently upgraded to 14.04, so I don't think I'll be upgrading to 15.10, as it is not a LTS release.
I'll try the newest version of GRIP under Windows Server 2008 R2 (which I will **not** upgrade) at some point soon.

FlamingSpork, 2016-02-01:
I installed v1.1.1 under Windows Server 2008 R2, but it gave me the same error as before.

JLLeitschuh, 2016-02-01:
Is this a 32 or 64 bit machine?

ThomasJClark, 2016-02-01:
> ```platform: /Linux/amd64/```

64 bit

PR #648 (2016-08-02): Add Launcher Binary
closes #566 
closes #345 
closes #370 

This solves the Linux build issues!

Thank you @PeterJohnson and @FibreFoX for your work on this.
closed 2016-08-02 by JLLeitschuh