WPIRoboticsProjects/GRIP #374: El Capitan crash

AlexPoulsen, 2016-01-16:
[image]
Crash on launch
10.11.2

PaulaRudy, 2016-01-18:
Hello there! Could you please describe the circumstances of the crash?

AlexPoulsen, 2016-01-18:
Opens, bounces, quits
About .5-5 seconds

PaulaRudy, 2016-01-18:
Do you happen to have a GRIP.log file in your home folder? (If so, could you please upload it?)

AlexPoulsen, 2016-01-18:
[GRIP.log.zip](https://github.com/WPIRoboticsProjects/GRIP/files/94979/GRIP.log.zip)

JLLeitschuh, 2016-01-18:
Try this: 
1. Go to your Library folder on your mac (how to here: http://www.macinstruct.com/node/551)
2. Then go to `Saved Application State` and see if there is a file that has something like `edu.wpi.grip`. The file name may be longer than that.
3. Delete/Trash that file.
4. Try to open GRIP again.

AlexPoulsen, 2016-01-18:
I've had all 3 Library folders in the sidebar for years
And nope it's not there

JLLeitschuh, 2016-01-18:
I'm running El Capitan and I've never seen this problem.
Someone else is experiencing this problem on windows. This one has me stumped. I have no idea how to reproduce it.

What version of gcc/g++ version do you have on your machine? In your terminal type
`gcc -v` and `-g++ -v`

How tech savvy are you?
Are you willing/able to download the source code for v1.0.0 to try attaching a debugger to GRIP when you launch it?

Baring that, try reinstalling.

AlexPoulsen, 2016-01-18:
(gcc -v)
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/c++/4.2.1
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.2.0
Thread model: posix
(and -g++ -v)
-bash: -g++: command not found

g++ -v returns same as gcc -v

redownloaded and it still crashed

I'm not all that good with compiling source code and don't know anything about attaching a debugger to GRIP

JacisNonsense, 2016-01-20:
Could we get some information on your system? "About My Mac" should provide everything. The OS X Version, Mac Model and CPU/GPU models should be enough for us to attempt to recreate the issue.

AlexPoulsen, 2016-01-20:
Accidentally updated to 10.11.3, it still crashes, but this time (after redownloading), it switches my gpu from Intel Iris Pro to Nvidia GeForce GT 750M then switches back after crashing
10.11.3
Intel Iris Pro/Nvidia GeForce GT 750M
16g ram
Retina/15in/Mid-2014
4 core 2.5 GHz core i7
anything else?

AlexPoulsen, 2016-01-20:
I don't think the GPU switched before

JacisNonsense, 2016-01-20:
That should be it. Thank you

JLLeitschuh, 2016-01-27:
Are you experiencing this problem with the newest release? v1.1.1

AlexPoulsen, 2016-01-27:
yes, I am
although it stops bouncing before crashing

JLLeitschuh, 2016-01-28:
This issue confounds me. I have no idea what is causing the problem.
I wish I had your computer in front of me to try to debug this problem.

Did you download and install the `grip-1.1.1-x64.dmg` or the `grip-1.1.1-x64.pkg` file?

AlexPoulsen, 2016-02-02:
Just downloaded again, and it still crashes

JLLeitschuh, 2016-02-02:
Are you willing to try building from source?
That may give us the information that we need to solve this problem for you.
Clone the repository from git and then follow the guide here:
https://github.com/WPIRoboticsProjects/GRIP/wiki/Setting-up-build-tools

Once you get it running tell us what the console outputs. There is probably some critical piece of info that we are missing.

AustinShalit, 2016-07-29:
@AlexPoulsen Are you still having issues with the [latest version](https://github.com/WPIRoboticsProjects/GRIP/releases) of GRIP?

AustinShalit, 2016-12-10:
@AlexPoulsen Please reopen this issue if you are still having trouble with the [latest release](https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.5.0).

closed 2016-12-10 by AustinShalit