WPIRoboticsProjects/GRIP #377: Fix deploying

ThomasJClark, 2016-01-16:
There's definitely something up with the deploy feature.  A lot of these reports have little useful information, but hopefully between them there should be enough to figure out what's going on.

- http://www.chiefdelphi.com/forums/showpost.php?p=1523851&postcount=49
- http://www.chiefdelphi.com/forums/showpost.php?p=1524008&postcount=54
- http://www.chiefdelphi.com/forums/showpost.php?p=1524508&postcount=62
- #352
- #367
- #375
- #376

bradamiller, 2016-01-16:
I wonder if there’s something that we can add to GRIP that helps diagnose deployment issues so that teams will have better information to report. Maybe more error checking around the ssh/sftp/scp connection that will provide some verbose output to see where it hangs up.

Brad

http://www.chiefdelphi.com/forums/showpost.php?p=1523851&postcount=49
http://www.chiefdelphi.com/forums/showpost.php?p=1524008&postcount=54
http://www.chiefdelphi.com/forums/showpost.php?p=1524508&postcount=62
#352
#367
#375
#376

AustinShalit, 2016-01-16:
I have yet to try to deploy to the RoboRIO.  I will try tomorrow to see if I run into any similar issues.

bigabos-hey, 2016-01-17:
We are having significant issues with deploy as well.  We've tested from 2 different laptops with 2 different roboRIOs.  Can't find a combo that works.
Most attempts to deploy get a "Failed to Connect".  Occasionally, and randomly, the deploy seems to work, but we get errors like I've pasted below.  
Really powerful tool guys.  If you want me to try anything, please holler.

An event subscriber threw an exception
## Stack Trace:
```java
java.util.concurrent.TimeoutException: Unable to terminate video feed from IP Camera 10.20.13.9
	at edu.wpi.grip.core.sources.CameraSource.stop(CameraSource.java:279)
	at edu.wpi.grip.core.sources.CameraSource.onStopPipeline(CameraSource.java:338)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.ui.DeployerController.onDeploy(DeployerController.java:107)
	at edu.wpi.grip.ui.deployment.DeploymentOptionsController.lambda$deploy$21(DeploymentOptionsController.java:81)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:107)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:98)
	at org.jdeferred.impl.DeferredObject.resolve(DeferredObject.java:70)
	at edu.wpi.grip.ui.deployment.FRCDeploymentOptionsController.lambda$onDeploy$25(FRCDeploymentOptionsController.java:71)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:107)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:98)
	at org.jdeferred.impl.DeferredObject.resolve(DeferredObject.java:70)
	at org.jdeferred.DeferredFutureTask.done(DeferredFutureTask.java:83)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:384)
	at java.util.concurrent.FutureTask.set(FutureTask.java:233)
	at java.util.concurrent.FutureTask.run(FutureTask.java:274)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.0.0
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 7
os.version | 6.1
os.arch | amd64

AaronPinto, 2016-01-20:
I am also having a problem deploying to the roboRIO. I can ping the roboRIO using mDNS or it's IP address on Wifi and Ethernet but I can't deploy to it.  I am using Windows 10.

closed 2016-01-20 by JLLeitschuh