WPIRoboticsProjects/GRIP #380: Problem when deploying to Roborio

sambattalio, 2016-01-16:
I currently have a problem when deploying code to the roborio with C++. I can get the GRIP files onto the rio from the GRIP deploy, and have checked their paths through ftp (I also have JRE), but cannot start GRIP through the code. When I place the line "execl(JAVA, "-jar", GRIP_JAR, GRIP_PROJECT, nullptr)" in my code and deploy to the roborio, the drivers station shows "no code". In eclipse it shows no errors, and I feel it might be a problem with my "deploy" of GRIP in my Roborio.  What should I do in order to fix this issue? 
Thanks!
(If needed I can provide more information that may be pertinent)

ThomasJClark, 2016-01-16:
What does your riolog say after you delploy in eclipse? (Window > Show View > Other > Riolog)

bradamiller, 2016-01-17:
Also, If it's a pretty simple java program you can post that and we can try to reproduce it.

Thanks!
Brad

sambattalio, 2016-01-17:
Sorry, I had to leave the Robotics lab for the day and cannot look at the riolog. I can get you the riolog tomorrow most likely. The program is with C++ using command based programming. It runs the execl() function in the constructor of one of the subsystems, that runs on init of the robot. I hope that helps for the time being, I will be able to give thorough information tomorrow, and if not tomorrow, definitely on monday. Thank you both for your responses, GRIP is really helpful so far :+1: .

sambattalio, 2016-01-18:
Error: Could not find or load main class .home.lvuser.grip.jar
 I have confirmed that grip.jar is on the roborio, in the right folder, through FTP. Is this a problem with our deploy of grip.jar to the roborio?

ThomasJClark, 2016-01-18:
That's strange.  Can you post a simple Java program to reproduce it?  Also, maybe get the grip.jar with FTP and upload it too

sambattalio, 2016-01-18:
This is in c++, here is a small program to showcase this error.  I have also attached the site for the grip.jar to this post. If you need anymore files let me know.
https://drive.google.com/file/d/0B1qdxHB3y8LCWERpR21VZXg4ejg/view?usp=sharing

```
#include <ErrorBase.h>
#include <networktables/NetworkTable.h>
#include <RobotBase.h>
#include <SampleRobot.h>
#include <Timer.h>
#include <unistd.h>
#include <iostream>
#include <memory>
#include <string>
#include <CameraServer.h>
#include <llvm/ArrayRef.h>
#include "WPILib.h"

/**
 * This is a demo program showing the use of the RobotDrive class.
 * The SampleRobot class is the base of a robot application that will automatically call your
 * Autonomous and OperatorControl methods at the right time as controlled by the switches on
 * the driver station or the field controls.
 *
 * WARNING: While it may look like a good choice to use for your code if you're inexperienced,
 * don't. Unless you know what you are doing, complex code will be much more difficult under
 * this system. Use IterativeRobot or Command-Based instead if you're new.
 */
class Robot: public SampleRobot
{
	const char * const JAVA = "/usr/local/frc/JRE/bin/java";
	const char * const GRIP_JAR = "/home/lvuser/grip.jar";
	const char * const GRIP_PROJECT = "home/lvuser/dankmemesv3.grip";
	std::unique_ptr<NetworkTable> grip;

public:
	Robot() {}

	void RobotInit()
	{
		grip.reset(NetworkTable::GetTable("GRIP").get());
		std::cout << "GRIP STARTED before" << std::endl;
		if (execl(JAVA,"-jar", GRIP_JAR, GRIP_PROJECT, nullptr) == -1) {
			std::cout << "GRIP didn't work.";
		}

		std::cout << "GRIP STARTED after" << std::endl;
	}

	void Autonomous() {}

	void OperatorControl()
	{
		while (IsOperatorControl() && IsEnabled()) {

			Wait(0.005);				// wait for a motor update time
		}
	}

	void Test() {}
};

START_ROBOT_CLASS(Robot)
```

ThomasJClark, 2016-01-18:
My bad, I make a mistake with the wiki example.  Try this:

```c++
const char *JAVA = "/usr/local/frc/JRE/bin/java";
char *GRIP_ARGS[5] = { "java", "-jar", "/home/lvuser/grip.jar",
                       "/home/lvuser/dankmemesv3.grip", NULL };
```

```c++
void RobotInit() override {
    if (fork() == 0) {
        if (execv(JAVA, GRIP_ARGS) == -1) {
            perror("Error running GRIP");
        }
    }
}
```

sambattalio, 2016-01-18:
Thanks! I will let you know if if works tomorrow. I unfortunately had to leave my school due to the weather.

sambattalio, 2016-01-19:
The line:
```
execv(JAVA, GRIP_ARGS);
```

is still crashing the program, and we get NO CODE on the roborio. 

In the riolog we get:

Jan 19, 2016 9:17:55 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.

OpenCV Error: Assertion failed (m.dims >= 2) in Mat, file /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp, line 441

WARNING: The CV dilate operation did not perform correctly.

SEVERE: The HSL Threshold operation did not perform correctly.
java.lang.IllegalArgumentException: HSL Threshold needs a 3-channel input

Any ideas what is causing this? Is there something we are missing? Thanks again for your continued help so far! :+1:

ThomasJClark, 2016-01-19:
Can I see your new robot program?

sambattalio, 2016-01-19:
```
#include <ErrorBase.h>
#include <networktables/NetworkTable.h>
#include <RobotBase.h>
#include <SampleRobot.h>
#include <Timer.h>
#include <unistd.h>
#include <iostream>
#include <memory>
#include <string>
#include <CameraServer.h>
#include <llvm/ArrayRef.h>
#include "WPILib.h"

/**
 * This is a demo program showing the use of the RobotDrive class.
 * The SampleRobot class is the base of a robot application that will automatically call your
 * Autonomous and OperatorControl methods at the right time as controlled by the switches on
 * the driver station or the field controls.
 *
 * WARNING: While it may look like a good choice to use for your code if you're inexperienced,
 * don't. Unless you know what you are doing, complex code will be much more difficult under
 * this system. Use IterativeRobot or Command-Based instead if you're new.
 */
class Robot: public SampleRobot
{
	char *const GRIP_ARGS[5] = {"java", "-jar",
			"/home/lvuser/grip.jar",
			"/home/lvuser/dankmemesv3.grip", NULL};
	const char *JAVA = "/usr/local/frc/JRE/bin/java";
	std::unique_ptr<NetworkTable> grip;

public:
	Robot() {}

	void RobotInit()
	{
		grip.reset(NetworkTable::GetTable("GRIP").get());
		std::cout << "GRIP STARTED before" << std::endl;
		execv(JAVA, GRIP_ARGS);
			//perror("Error running GRIP");

		std::cout << "GRIP STARTED after" << std::endl;
	}

	void Autonomous() {}

	void OperatorControl()
	{
		while (IsOperatorControl() && IsEnabled()) {

			Wait(0.005);				// wait for a motor update time
		}
	}

	void Test() {}
};

START_ROBOT_CLASS(Robot)
```

sambattalio, 2016-01-19:
The perror is commented just because of testing we were doing to look for the exact line throwing the error.

ThomasJClark, 2016-01-19:
`execv()` replaces the current process, which is why you're getting the "No Code" thing.  You need to `fork()` first.

```c
if (fork() == 0) {
    if (execv(JAVA, GRIP_ARGS) == -1) {
        perror("Error running GRIP");
    }
}
```

sambattalio, 2016-01-19:
Oh! My mistake, it seems to be working so far now! Thanks so much :+1:

ThomasJClark, 2016-01-19:
no problem

sambattalio, 2016-01-19:
Sorry! One more thing, is this anything that we can fix? Or what is the cause of this?

#
# There is insufficient memory for the Java Runtime Environment to continue.
# Native memory allocation (mmap) failed to map 32768 bytes for committing reserved memory.
# An error report file with more information is saved as:
# /tmp/hs_err_pid2625.log
#

This is in the riolog. Thanks :+1:

ThomasJClark, 2016-01-19:
It's possible there are somehow multiple instances of GRIP running. Try restarting your RIO, or (if you can) run `killall java` in an SSH session

sambattalio, 2016-01-20:
Is there any way to check in code if GRIP is already running? Just wondering because that would help out a lot :smile_cat:

ThomasJClark, 2016-01-20:
I guess you could do this:

```c
if (system("pidof java") != 0) {
    if (fork() == 0 && execv(JAVA, GRIP_ARGS) == -1) {
        perror("Error running GRIP");
    }
}
```

sambattalio, 2016-01-20:
Thanks, that helped solve that problem. Thanks to your help, we can initialize grip without the robot code crashing.

 I was having troubles with the Network tables though, I have the error:

NT: ERROR: bind() failed: Address already in use (TCPAcceptor.cpp:102)
1559

I know that TCPAcceptor is not a part of GRIP, but as far as I can tell, GRIP is the culprit for this error.

Along with that, the robot completely crashes when I add the line:

```
areas = grip->GetNumberArray("myContoursReport/area", llvm::ArrayRef<double>());		
```

I believe this is crashing due to the previous error.

JLLeitschuh, 2016-01-27:
I'm going to close this issue because it seems to be resolved. If not please feel free to reopen it.

closed 2016-01-27 by JLLeitschuh