WPIRoboticsProjects/GRIP #381: Installed IntelliJ to get the latest

cpapplefamily, 2016-01-17:
I installed IntelliJ to try and get the latest n greatest. When I try to run it the build fails with a about 100 errors

\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
10:25:20 PM: Executing external task ':ui:run'...
:buildSrc:compileJava UP-TO-DATE
:buildSrc:extractSources UP-TO-DATE
:buildSrc:compileGroovy UP-TO-DATE
:core:shadowJar UP-TO-DATE
C:\Users\Applefamily\Documents\Corey's\Intellij\GRIP-master\GRIP-master\ui\src\main\java\edu\wpi\grip\ui\DeployerController.java:29: error: cannot find symbol
    private DialogPane root;
            ^
  symbol:   class DialogPane
  location: class DeployerController
C:\Users\Applefamily\Documents\Corey's\Intellij\GRIP-master\GRIP-master\ui\src\main\java\edu\wpi\grip\ui\DeployerController.java:139: error: cannot find symbol
    public DialogPane getRoot() {
           ^
  symbol:   class DialogPane
  location: class DeployerController
C:\Users\Applefamily\Documents\Corey's\Intellij\GRIP-master\GRIP-master\ui\src\main\java\edu\wpi\grip\ui\deployment\FRCDeploymentOptionsController.java:9: error: cannot find symbol
import javafx.scene.control.Spinner;
                           ^
  symbol:   class Spinner
  location: package javafx.scene.control
C:\Users\Applefamily\Documents\Corey's\Intellij\GRIP-master\GRIP-master\ui\src\main\java\edu\wpi\grip\ui\deployment\FRCDeploymentOptionsController.java:10: error: cannot find symbol
import javafx.scene.control.SpinnerValueFactory;
                           ^
...
...
... and so on
...
...

symbol:   class SpinnerValueFactory
  location: class ListSpinnerInputSocketController
Note: C:\Users\Applefamily\Documents\Corey's\Intellij\GRIP-master\GRIP-master\ui\src\main\java\edu\wpi\grip\ui\util\deployment\DeployedInstanceManager.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
100 errors
:ui:compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':ui:compileJava'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 7.184 secs
Compilation failed; see the compiler error output for details.
10:25:30 PM: External task execution finished ':ui:run'.
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\

seems to be missing symbols and parts for the gui. Any insight?

ThomasJClark, 2016-01-17:
Can you post the output of  `java -version` and `javac -version`?

cpapplefamily, 2016-01-17:
I got it to work today while at work (on break) on my laptop. The trouble I was having was on my PC at home. I don't recall doing a step on the PC that I was forced to complete this am on the laptop.

Setting Project defaults for the Project SDK to 1.8

The Wiki says to do this and while importing on the PC the Gradle JVM did show 1.8. As for now I'm not even sure how to find that menu from the workspace. When I did set it up it was from the start up splash screen that no longer opens.

Anyhow cant wait to give this a test burn. I see a lot of new publish blocks and M-JPEG Streamer.

PaulaRudy, 2016-01-18:
Hey there! Sorry to here you're having issues setting up GRIP with intellij. First off, what version of java do you have installed? You can find instructions to find this information [here]https://java.com/en/download/help/version_manual.xml).

closed 2016-01-21 by ThomasJClark