WPIRoboticsProjects/GRIP #385: M-JPEG Stream as input to Grip on PC

cpapplefamily, 2016-01-18:
I think it would be helpful if you could get the M-JPEG Stream from the robot to Grip running on the PC. I suppose you could run grip.jar with a small profile.grip only streaming the video and not processing it. Then have a block to read image from Network Tables.  

1 A few benefits would be leaving the camera mounted to the Robot and using its mounted position for set up.
2 Using the camera that the profile.grip will be deployed with.

3 Running the vision algorithm off robot. Not sure how much CPU this would save.

JLLeitschuh, 2016-01-18:
You can do this currently if you have your computer and the network camera plugged into the dlink. Unless I'm not understanding your request correctly.

cpapplefamily, 2016-01-18:
We are using a USB camera. I did mention to get an ip camera. I would like to know if there are performance benefits to one over the other.

bradamiller, 2016-01-18:
Using an IP camera gives you the opportunity to send the images directly back to the driver station where you can run GRIP without having to send the video stream through the roboRIO. This makes the robot program able to run at fill speed all the time and not be effected by the vision system. The GRIP program can send the results back to the robot via NetworkTables.

ThomasJClark, 2016-01-18:
Yeah, the roboRIO isn't nearly powerful enough to serve 24-30 frames per second.  If you're not processing images on the rio, it's probably a good idea to avoid having them go through it.

cpapplefamily, 2016-01-18:
All good to know with a first year vision team.

bradamiller, 2016-01-18:
It's not that it can't work, and lots of teams do it with lower res images. In either case if you're using the camera for aiming, don't have a PID loop where the image coordinates are the source. Instead, use the camera to generate set points and use some other much faster sensor drive your aiming.

ThomasJClark, 2016-01-19:
It looks like at least two other teams want support for the dashboard protocol as an input: http://www.chiefdelphi.com/forums/showthread.php?t=141952

This is not the most efficient way to do vision processing, but it's probably the cheapest in terms of hardware, and it shouldn't be too much trouble to add support for it in GRIP

cpapplefamily, 2016-01-19:
Now that I know how much of a disappointment usb cam steem to pc will be we are going to avoid it. I do like the fact I might be able to draw a custom mask over the video feed to the smart dashboard for aiming manually. Either pc driven or maybe rio.

SamCarlberg, 2016-11-18:
This is no longer a problem. WPILib is adding a camera server API for 2017 to let the robot stream images from USB cameras. See [the cscore project](https://github.com/wpilibsuite/cscore).

closed 2016-11-18 by SamCarlberg