WPIRoboticsProjects/GRIP #387: GRIP IP Camera Stream Crashing

bhavinshahteam696, 2016-01-19:
Hello everyone,

I've been toying around with GRIP for the last week and I have to say it's an incredibly useful program, especially for Stronghold. However, our team has run into a problem where input from an Axis IP camera randomly crashes after a short time duration (anywhere from a few seconds to a few minutes). Sometimes, GRIP only loads up the first frame of the stream before freezing up. When I try to end the video stream and restart it, GRIP throws an exception (cannot stop IP camera stream) before crashing.

I know for a fact that the IP camera is working fine as I can view the video stream in a browser with the IP of the camera. I've tried restarting GRIP, uninstalling and reinstalling the stable release, restarting my computer, and using a completely different computer. I've attached my log and the GRIP exception error I mentioned earlier. 

Any help would be greatly appreciated.

TECHNICAL DETAILS:
OS: Windows 7
Release: Stable exe_x64
Camera: Axis M1013
Setup: Axis camera connected via ethernet to RoboRIO. Laptop running GRIP connected over Wifi to RoboRIO. Firewall disabled.
[GRIP_log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/95169/GRIP_log.txt)
[Exception_Error.txt](https://github.com/WPIRoboticsProjects/GRIP/files/95170/Exception_Error.txt)

setupminimal, 2016-01-25:
We are also having this issue, with exactly the same setup. This is pretty important to our team, so if you need help testing or replicating it, I would be happy to help.

nardavin, 2016-01-26:
Same issue here. Crash log:

An event subscriber threw an exception
## Stack Trace:
```java
java.util.concurrent.TimeoutException: Unable to terminate video feed from IP Camera 10.24.3.11
	at edu.wpi.grip.core.sources.CameraSource.stop(CameraSource.java:279)
	at edu.wpi.grip.core.sources.CameraSource.onSourceRemovedEvent(CameraSource.java:312)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.ui.pipeline.source.SourceController.delete(SourceController.java:103)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor3.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1771)
	at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1657)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$ClickGenerator.postProcess(Scene.java:3470)
	at javafx.scene.Scene$ClickGenerator.access$8100(Scene.java:3398)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3766)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.1.1
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 7
os.version | 6.1
os.arch | amd64

PR #456 (2016-02-05): Fixes IP Cameras Hanging
 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Related bytedeco/javacv#322
commit dbc156d1 (2016-02-05): Fixes IP Cameras Hanging

 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Closes #460
Related https://github.com/bytedeco/javacv/issues/322
closed 2016-02-05 by ThomasJClark