WPIRoboticsProjects/GRIP #388: I don't know what happened...

kaitoa19, 2016-01-19:
Thread[JavaFX Application Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.NullPointerException
	at edu.wpi.grip.ui.DeployerController.lambda$null$14(DeployerController.java:133)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.0.0
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Mac OS X
os.version | 10.11.3
os.arch | x86_64

ThomasJClark, 2016-01-19:
See issue #386 for all deploy-related UI problems

closed 2016-01-19 by ThomasJClark

ThomasJClark, 2016-01-22:
[v1.1.0](https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.1.0) is out - this should fix most deploying problems