WPIRoboticsProjects/GRIP #389: Gradle from source invalid source release

ajfreed00, 2016-01-19:
So the problem is basically this right here. I'm running it on a Raspberry Pi 2 Model B with Raspbian Jessie. What am I doing wrong?

Note: in order to honour the org.gradle.jvmargs and/or org.gradle.java.home values specified for this build, it is necessary to fork a new JVM.
To avoid the slowdown associated with this extra process, you might want to consider running Gradle with the daemon enabled.
Please see the user guide chapter on the daemon at http://gradle.org/docs/1.5/userguide/gradle_daemon.html.
:buildSrc:clean
:buildSrc:compileJava
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:buildSrc:extractSources UP-TO-DATE
:buildSrc:compileGroovy
:buildSrc:processResources FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':processResources'.
> Could not copy file '/home/pi/Downloads/GRIP-1.0.0/buildSrc/src/main/resources/META-INF' to '/home/pi/Downloads/GRIP-1.0.0/buildSrc/build/resources/main/META-INF'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 5 mins 6.102 secs

ajfreed00, 2016-01-19:
Oops, the title is wrong on this. xD

JLLeitschuh, 2016-01-19:
JavaFX does not work on a PI because it is not included in ARM Java.

ajfreed00, 2016-01-19:
Oh ok thanks. Does GRIP work at all on any Windows 32 bit systems?

JLLeitschuh, 2016-01-19:
Yes, there are downloads on the release page.
You can also build from source if you wish and it should work as well.

For anyone who wants to work on this in the future we may be able to provide a release that uses the openJDK for a pi release.
https://www.linkedin.com/pulse/oracle-just-removed-javafx-support-arm-jan-snelders
I have no idea how complicated that would be.
I'm sure we could probably get travis to build with the openJDK automatically.

ajfreed00, 2016-01-19:
Well, there is a bit of another problem... Whenever I try to install the
last "stable" release of GRIP, the installer says my Windows is not
compatible. I'm running Windows 10 x32, is that a problem?

ThomasJClark, 2016-01-19:
> Well, there is a bit of another problem... Whenever I try to install the
> last "stable" release of GRIP, the installer says my Windows is not
> compatible. I'm running Windows 10 x32, is that a problem?

Which installer are you using?  If you're not on a 64 bit machine, make sure it's `grip-1.0.0-x86.exe`

ajfreed00, 2016-01-19:
Lol, I don't know if something got changed or what, but the first Windows
installer didn't say x86 OR x64. It was just grip-1.0.0.exe. xD Now I'm
installing the RIGHT one. Lol thanks for your help, guys!

ThomasJClark, 2016-01-19:
Yeah, there was originally only a 64 bit build.  I added the `-x86` and `-x64` suffixes later

JLLeitschuh, 2016-01-27:
I think this is a duplicate of #366
I'm going to close this issue and direct people to that issue in the future.

closed 2016-01-27 by JLLeitschuh