WPIRoboticsProjects/GRIP #390: CameraSource Error

AaronPinto, 2016-01-20:
IP Camera 10.13.25.23 Frame Grabber Thread crashed with uncaught exception

Stack Trace

java.lang.NullPointerException
	at edu.wpi.grip.core.sources.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:137)
	at edu.wpi.grip.core.sources.CameraSource.lambda$start$18(CameraSource.java:200)
	at java.lang.Thread.run(Thread.java:745)

I also tried using this URL as well
http://10.13.25.23/mjpg/video.mjpg

JLLeitschuh, 2016-01-20:
Can you open that address in your web browser?

AaronPinto, 2016-01-20:
Yes I can view the video stream in my web browser.

JLLeitschuh, 2016-01-21:
I've opened an issue with the library author. 
https://github.com/bytedeco/javacv/issues/312
We can fix it locally in our version too.

AaronPinto, 2016-01-21:
Great thank you! :D

PR #456 (2016-02-05): Fixes IP Cameras Hanging
 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Related bytedeco/javacv#322
commit dbc156d1 (2016-02-05): Fixes IP Cameras Hanging

 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Closes #460
Related https://github.com/bytedeco/javacv/issues/322
closed 2016-02-05 by ThomasJClark