WPIRoboticsProjects/GRIP #393: PublishContoursReport only uses Rectangles, not Quadrilaterals

Poppeseed, 2016-01-20:
Since the Publish ContoursReport operation uses the boundingRect, any time you are not facing the target directly head on the report is a bit inaccurate. I'm not well-versed in OpenCV, but it seems the only way to currently get this sort of information is the Publish LinesReport operation, or a combination of the two.

Is it possible to add a new Publish ContoursReport operation that bounds using a quadrilateral instead of a rectangle?

Poppeseed, 2016-01-21:
Looking into it a bit further, if you could publish a report of the actual points after a ConvexHull contour, it would likely solve this.

[image]

ThomasJClark, 2016-01-21:
Yeah, I don't think there's a simple, efficient solution for finding bounding quadrilaterals.  Publishing a list of points might be good, but I don't think NetworkTables supports 2D arrays, and even if it did array sizes are limited to 255 elements.

Maybe we can extract a *single* contour at a time and publish up to 255 x/y coords?

Poppeseed, 2016-01-21:
Yeah, without 2D arrays you'd be pretty limited. Having a criteria selection available (min/max area/width/height/etc) would probably work.

ThadHouse, 2016-01-21:
NetworkTables 3 can also send raw data, so GRIP could create their own raw data type for sending more complicated arrays. I would require code on the team end to decode, but at least the teams could get the data.