WPIRoboticsProjects/GRIP #394: Resouce Leak warning in Eclipse

rlee287, 2016-01-20:
Eclipse points out two resource leaks:
```Resource leak: '\<unassigned Closeable value\>' is never closed
FindBlobsOperation.java
/core/src/main/java/edu/wpi/grip/core/operations/composite
line 68```
And
```Resource leak: 'expectedResult' is never closed
AddOperationTest.java
/core/src/test/java/edu/wpi/grip/core/operations/opencv
line 72```
Also, after building with Gradle in Eclipse using `:ui:run` and closing GRIP, two `java.exe` processes are left behind in Task Manager, both taking up at least 256 MB of memory. I am not sure whether this is related, but I am putting this here just in case.

JLLeitschuh, 2016-01-20:
GRIP uses gradle which (if you have daemon mode enabled) will keep gradle in memory to speed up a gradle task in the future.
I don't know if that's whats causing the the java program to stick around.

ThomasJClark, 2016-01-21:
The resource leak thing I believe is a non-issue.  JavaCV objects all implement `AutoClosable`, which lets you deallocate the native C++ objects with a method call.  However, they will still deallocate themselves when garbage collected.

JavaCV is kind of weird.

closed 2016-01-21 by ThomasJClark