WPIRoboticsProjects/GRIP #398: Error thrown during resize

FRC-3277, 2016-01-21:
The application thew an error when resize of image fx and fx set to 4 and the preview socket button clicked. 

An event subscriber threw an exception
## Stack Trace:
```java
java.lang.IndexOutOfBoundsException: Index: 1, Size: 1
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.sun.javafx.collections.VetoableListDecorator.get(VetoableListDecorator.java:306)
	at javafx.scene.Parent$2.onProposedChange(Parent.java:423)
	at com.sun.javafx.collections.VetoableListDecorator.remove(VetoableListDecorator.java:329)
	at edu.wpi.grip.ui.preview.PreviewsController.lambda$onSocketPreviewChanged$115(PreviewsController.java:173)
	at edu.wpi.grip.ui.util.GRIPPlatform.runAsSoonAsPossible(GRIPPlatform.java:64)
	at edu.wpi.grip.ui.preview.PreviewsController.onSocketPreviewChanged(PreviewsController.java:143)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.OutputSocket.setPreviewed(OutputSocket.java:53)
	at edu.wpi.grip.ui.pipeline.OutputSocketController.lambda$initialize$93(OutputSocketController.java:73)
	at com.sun.javafx.binding.ExpressionHelper$SingleInvalidation.fireValueChangedEvent(ExpressionHelper.java:137)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.BooleanPropertyBase.fireValueChangedEvent(BooleanPropertyBase.java:103)
	at javafx.beans.property.BooleanPropertyBase.markInvalid(BooleanPropertyBase.java:110)
	at javafx.beans.property.BooleanPropertyBase.set(BooleanPropertyBase.java:144)
	at javafx.scene.control.ToggleButton.setSelected(ToggleButton.java:149)
	at javafx.scene.control.ToggleButton.fire(ToggleButton.java:255)
	at com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(ButtonBehavior.java:182)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:96)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:89)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.0.0
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 7
os.version | 6.1
os.arch | amd64

JLLeitschuh, 2016-01-21:
It seems that it is possible for the ordering of the sockets to get out of sync with the ordering of the backing list.

PaulaRudy, 2016-01-21:
@JLLeitschuh  What makes you think this is an issue with the ordering of the previews?

ThomasJClark, 2016-01-21:
It's definitely a bug in the preview ordering code.  There was a NullPointerException on [this](https://github.com/WPIRoboticsProjects/GRIP/blob/master/ui/src/main/java/edu/wpi/grip/ui/preview/PreviewsController.java#L173) line.

Instead of trying to fix it, I really recommend using [`FXCollections.sort`](https://docs.oracle.com/javase/8/javafx/api/javafx/collections/FXCollections.html#sort-javafx.collections.ObservableList-java.util.Comparator-)

PaulaRudy, 2016-01-21:
I can't seem to reproduce this error. I will implement FXCollections.sort if you think it is necessary, however.

PR #406 (2016-01-22): Use FXCollections.sort to sort preview views
Instead of manually inserting them, which is verbose and error prone, we
can use JavaFX's built-in sorting routines to make the preview order
match up with the order of the things in the pipeline being previewed

This be merged in before we do the release, since the old code causes crashes

Closes #398
commit f24c08dc (2016-01-22): Use FXCollections.sort to sort preview views

Instead of manually inserting them, which is verbose and error prone, we
can use JavaFX's built-in sorting routines to make the preview order
match up with the order of the things in the pipeline being previewed

Closes #398
closed 2016-01-22 by ThomasJClark