WPIRoboticsProjects/GRIP #400: GRIP cannot load certain filenames

Short-Circuit, 2016-01-21:
This error occurred while attempting to add an image source named "ss+(2016-01-21+at+07.57.26).jpg"
It appears that the name is being improperly encoded as an HTML URL (special escape character "+" is displayed as a space " " in the program). The issue resolved itself when the image was renamed to not contain special characters.

```
The image selected was invalid
## Stack Trace:
```java
java.io.IOException: Error loading image $HOME\Desktop\ss (2016-01-21 at 07.57.26).jpg
	at edu.wpi.grip.core.util.ImageLoadingUtility.loadImage(ImageLoadingUtility.java:35)
	at edu.wpi.grip.core.sources.ImageFileSource.loadImage(ImageFileSource.java:118)
	at edu.wpi.grip.core.sources.ImageFileSource.loadImage(ImageFileSource.java:113)
	at edu.wpi.grip.core.sources.ImageFileSource.initialize(ImageFileSource.java:86)
	at edu.wpi.grip.ui.pipeline.AddSourceView.lambda$new$49(AddSourceView.java:120)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$ClickGenerator.postProcess(Scene.java:3470)
	at javafx.scene.Scene$ClickGenerator.access$8100(Scene.java:3398)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3766)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.0.0
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 10
os.version | 10.0
os.arch | amd64
```

JLLeitschuh, 2016-01-21:
Thank you for reporting this. We will have to explore this more.

SamCarlberg, 2016-10-31:
This still happens in the latest version. It's caused by [this constructor]9https://github.com/WPIRoboticsProjects/GRIP/blob/master/core/src/main/java/edu/wpi/grip/core/sources/ImageFileSource.java#L54-L55) parsing the file name as a URI.

@ThomasJClark What was the reasoning behind this?

JLLeitschuh, 2016-10-31:
Probably so that it could be stored as a property when the file was deserialized. Either that or it was because that's what the openCV bindings expected as an input.

SamCarlberg, 2016-10-31:
The properties don't care, it's a String-to-String mapping. And I changed it on my machine to `this(..., ..., file.getAbsolutePath())` and OpenCV was fine with it

JLLeitschuh, 2016-10-31:
Sure, if it works go ahead.

PR #697 (2016-11-02): Don't parse image paths as URLs
Fixes #400
commit da32d697 (2016-11-02): Don't parse image paths as URLs (#697)

Fixes #400
closed 2016-11-02 by SamCarlberg