WPIRoboticsProjects/GRIP #403: Error when deploying with static images

rlee287, 2016-01-21:
This is the error log I got when deployed with a static image. I will attach the .grip file.
[Lines_samplepics.txt](https://github.com/WPIRoboticsProjects/GRIP/files/100070/Lines_samplepics.txt)

```
Loading Dependency Injection Framework
Sep 16, 2015 12:33:26 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Linux/arm/
Sep 16, 2015 12:33:37 AM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
Sep 16, 2015 12:33:40 AM edu.wpi.grip.core.Source initializeSafely
WARNING: Failed to initialize ImageFileSource
java.io.IOException: Error loading image C:\Users\Ryan\Downloads\FIRSTStronghold\RealFullField\364.jpg
	at edu.wpi.grip.core.util.ImageLoadingUtility.loadImage(ImageLoadingUtility.java:35)
	at edu.wpi.grip.core.sources.ImageFileSource.loadImage(ImageFileSource.java:118)
	at edu.wpi.grip.core.sources.ImageFileSource.loadImage(ImageFileSource.java:113)
	at edu.wpi.grip.core.sources.ImageFileSource.initialize(ImageFileSource.java:86)
	at edu.wpi.grip.core.Source.initializeSafely(Source.java:98)
	at edu.wpi.grip.core.serialization.SourceConverter.unmarshal(SourceConverter.java:60)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1190)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:68)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)
	at edu.wpi.grip.core.Main.start(Main.java:59)
	at edu.wpi.grip.core.Main.main(Main.java:41)

Sep 16, 2015 12:33:40 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Failed to initialize ImageFileSource
java.io.IOException: Error loading image C:\Users\Ryan\Downloads\FIRSTStronghold\RealFullField\364.jpg
	at edu.wpi.grip.core.util.ImageLoadingUtility.loadImage(ImageLoadingUtility.java:35)
	at edu.wpi.grip.core.sources.ImageFileSource.loadImage(ImageFileSource.java:118)
	at edu.wpi.grip.core.sources.ImageFileSource.loadImage(ImageFileSource.java:113)
	at edu.wpi.grip.core.sources.ImageFileSource.initialize(ImageFileSource.java:86)
	at edu.wpi.grip.core.Source.initializeSafely(Source.java:98)
	at edu.wpi.grip.core.serialization.SourceConverter.unmarshal(SourceConverter.java:60)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
[... 592 lines cut ...]
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1190)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:68)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)
	at edu.wpi.grip.core.Main.start(Main.java:59)
	at edu.wpi.grip.core.Main.main(Main.java:41)

Sep 16, 2015 12:33:41 AM edu.wpi.grip.core.Main start
INFO: SUCCESS! The project is running in headless mode!

```

EdWard680, 2016-01-21:
I believe Image Sources load from a file path, and when you deploy that path stays the same, it doesn't copy the images over as well. You can see that it's trying to find the picture in the line `C:\Users\Ryan\Downloads\FIRSTStronghold\RealFullField\364.jpg` but since its running on the Rio that file path doesn't exist. You would have to download the image to the rrio, and then set the image source to something like /home/lvuser/path/to/image

JLLeitschuh, 2016-01-22:
In the future we plan to create a save file bundle that we will upload with any image files you have imported. At present, deploying static images isn't supported out of the box.

closed 2017-01-08 by SamCarlberg