WPIRoboticsProjects/GRIP #404: Axis IP Camera can be added, but no Image seen.

gerth2, 2016-01-22:
Using GRIP 1.0.0 on a x64 windows system. Can view webcam live view in Chrome. Putting in the IP address of the camera seems to add it in GRIP, but no image preview is available. Framerate is reported at zero. No downstream connected blocks have any visual output. 

 Exception thrown when deleting the camera in GRIP:

An event subscriber threw an exception
## Stack Trace:
```java
java.util.concurrent.TimeoutException: Unable to terminate video feed from IP Camera 10.17.36.12
	at edu.wpi.grip.core.sources.CameraSource.stop(CameraSource.java:279)
	at edu.wpi.grip.core.sources.CameraSource.onSourceRemovedEvent(CameraSource.java:312)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.ui.pipeline.source.SourceController.delete(SourceController.java:103)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor3.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1771)
	at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1657)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$ClickGenerator.postProcess(Scene.java:3470)
	at javafx.scene.Scene$ClickGenerator.access$8100(Scene.java:3398)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3766)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.0.0
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 7
os.version | 6.1
os.arch | amd64

This issue is most likely downstream of the actual problem. Any good way to go about debugging why the image preview is not displaying in GRIP to start?

gerth2, 2016-01-22:
Looks like this might be similar or the same as #391 .

JLLeitschuh, 2016-01-22:
What did you enter when you created the webcam?
Make sure you don't have a trailing slash when creating the camera's IP address. We automatically append `/mjpg/video.mjpg` to the end of the URL if you don't do it.
Alternatively, try using the IP address `10.17.36.12/mjpg/video.mjpg` and see if that works.

gerth2, 2016-01-22:
Just attempted with:

   http://10.17.36.12/mjpg/video.mjpg , same behavior. Attempted to just enter "10.17.36.12" - http:// was automatically prepended within the dialog box, but still same behavior.

cyberx696, 2016-01-22:
Same issue as #387

gerth2, 2016-01-22:
Cool! We're not the only ones. I was never able to get any frames off of
our camera, but the exception on close looks the same.

JLLeitschuh, 2016-01-22:
I don't have access to a windows machine to test this bug on. Is there anyone out there not on a Windows machine that is also experiencing this problem?

EdWard680, 2016-01-23:
I'm experiencing this on Ubuntu 14.04 with a non-axis MPJEG stream. (Streaming from my raspi cam)

gerth2, 2016-01-26:
Tested on Windows x64 with GRIP 1.1.0 and 1.1.1 - same issue.

gerth2, 2016-01-26:
Sorry, punched the wrong button. Doing some more sampling now, hopefully will have something useful to report soon.

gerth2, 2016-01-26:
Ok, another sample point - tried on a different PC running windows 10. 

Were able to get some camera images sporadically - image stream would either start and work fine, or not start at all. Seemed that sometimes it would work the very first time after a fresh grip session was started and the camera had just been reset, but the break on subsequent IP camera deletion/add cycles within the same session of Grip. Other times, it would not display anything on the first IP camera add, produced the above stack trace on deletion,  then succeed displaying the stream on the second attempt to add an IP camera.

I'm leaning toward there being something in the frame grabber thread which isn't going 100% as expected, then on deletion the thread hangs and triggers the timeout. What is causing the hang isn't immediately obvious though....

imdunne8, 2016-01-26:
I was working with @gerth2 on this last night and we did some more testing on this.  It looks like we had something on the roboRio that was taking over the frame grabber at some times.  We disconnected the roboRio and had a 100% success rate on the Windows 10 machine assuming we didn't have an existing process accessing the frame grabber.  We believe the issue with the Windows 7 machines relates to some security software/settings installed by our sponsor.  We'll have a completely different (store purchased) Windows 7 laptop to try tonight to confirm that theory.  Assuming that all checks out it may make sense to close this issue, but if somebody wants us to dig into the source code and see where the error is popping up on our current Windows 7 machines we can try to figure that out when we have the time.

For others reporting similar issues, here's our setup that worked reliably every time in GRIP that you should be able to use as a baseline:
- Axis IP Camera attached to wireless router
- roboRIO disconnected from router or powered off
- Windows 10 machine running GRIP 1.1.1 connected wireless to router
- Power cycle camera and wait for ring to turn green (kills existing connections to framegrabber)
- In GRIP, Add IP Camera.  Type just the camera's IP address (http:// is appended automatically)
- Press OK (/mjpg/video.mjpg will be appended by GRIP)
- Click the eye icon to view the image from the webcam in GRIP

PR #456 (2016-02-05): Fixes IP Cameras Hanging
 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Related bytedeco/javacv#322
commit dbc156d1 (2016-02-05): Fixes IP Cameras Hanging

 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Closes #460
Related https://github.com/bytedeco/javacv/issues/322
closed 2016-02-05 by ThomasJClark