WPIRoboticsProjects/GRIP #405: GRIP crashes upon startup

ocurr, 2016-01-22:
When I run GRIP from robot code (in c++) I sometimes get the following error message:
```
NT: server: client CONNECTED: 10.0.0.144 port 59424
Loading Dependency Injection Framework
Jan 22, 2016 2:10:11 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Linux/arm/
Jan 22, 2016 2:10:16 AM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/myproject.grip
NT: server: client CONNECTED: 127.0.0.1 port 45296
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
INFO: ntcore/src/Dispatcher.cpp:392 client: CONNECTED to server 0.0.0.0 port 1735
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
OpenCV Error: Assertion failed (ssize.area() > 0) in resize, file /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp, line 3208
java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp:3208: error: (-215) ssize.area() > 0 in function resize

	at org.bytedeco.javacpp.opencv_imgproc.resize(Native Method)
	at edu.wpi.grip.generated.opencv_imgproc.Resize.perform(Resize.java:59)
	at edu.wpi.grip.core.Operation.perform(Operation.java:61)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:137)
	at edu.wpi.grip.core.Step.onInputSocketChanged(Step.java:156)
	at sun.reflect.GeneratedMethodAccessor2.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:483)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.serialization.ConnectionConverter.unmarshal(ConnectionConverter.java:58)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1190)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:68)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)
	at edu.wpi.grip.core.Main.start(Main.java:53)
	at edu.wpi.grip.core.Main.main(Main.java:35)
Jan 22, 2016 2:10:18 AM edu.wpi.grip.core.Main onExceptionClearedEvent
INFO: Exception Cleared Event
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGFPE (0x8) at pc=0xb6ef2ab8, pid=2088, tid=3063022688
#
# JRE version: Java(TM) SE Embedded Runtime Environment (8.0_06-b23) (build 1.8.0_06-b23)
# Java VM: Java HotSpot(TM) Embedded Client VM (25.6-b23 mixed mode linux-arm )
# Problematic frame:
# C  [libpthread.so.0+0x10ab8]  raise+0x24
#
# Core dump written. Default location: //core or core.2088 (max size 2048 kB). To ensure a full core dump, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /tmp/hs_err_pid2088.log
```
This seems to happen randomly because it will sometimes return success. This will also sometimes happen when I try to run GRIP directly after deploying it through the UI.

JLLeitschuh, 2016-01-22:
Can you also provide a copy of the `/tmp/hs_err_pid2088.log` file on your robot?

ocurr, 2016-01-22:
Sure:
```
#                                                                                         
# A fatal error has been detected by the Java Runtime Environment:                        
#                                                                                         
#  SIGFPE (0x8) at pc=0xb6f85ab8, pid=1547, tid=3063624800                                
#                                                                                         
# JRE version: Java(TM) SE Embedded Runtime Environment (8.0_06-b23) (build 1.8.0_06-b23) 
# Java VM: Java HotSpot(TM) Embedded Client VM (25.6-b23 mixed mode linux-arm )           
# Problematic frame:                                                                      
# C  [libpthread.so.0+0x10ab8]  raise+0x24                                                
#                                                                                         
# Core dump written. Default location: //core or core.1547 (max size 2048 kB). To ensure a
#                                                                                         
# If you would like to submit a bug report, please visit:                       
#   http://bugreport.sun.com/bugreport/crash.jsp                                          
# The crash happened outside the Java Virtual Machine in native code.                     
# See problematic frame for where to report the bug.                                      
#                                                                                         
                                                                                          
---------------  T H R E A D  ---------------                                             
                                                                                
Current thread (0xb6805800):  JavaThread "main" [_thread_in_native, id=1548, stack(0xb6964
                                                                               
siginfo:si_signo=SIGFPE: si_errno=0, si_code=-6 (ILL_ILLADR), si_addr=0x0000060b

Registers:                                                                                
  r0  = 0x00000000                                                                        
  r1  = 0x0000060c                                                                        
  r2  = 0x00000008                                                                        
  r3  = 0xb69b3920                                                                        
  r4  = 0x00000001                                                                        
  r5  = 0x00000000                                                                        
  r6  = 0xb69b2208                                                                        
  r7  = 0x0000010c                                                                        
  r8  = 0x00010000                                                                        
  r9  = 0x00010000                                                                        
  r10 = 0x00000000                                                              
  fp  = 0xb69b21d8                                                                        
  r12 = 0xb6f85a94                                                                        
  sp  = 0xb69b1b6c                                                                        
  lr  = 0xb6e0ff04                                                                        
  pc  = 0xb6f85ab8                                                                        
  cpsr = 0x20070010                                                                       
                                                                                          
Top of Stack: (sp=0xb69b1b6c)                                                             
0xb69b1b6c:   00000000 00000000 ad0affd0 ae2b1a38                               
0xb69b1b7c:   b6805800 b69b1bfc b6f29d54 00000000 
xb69b1b8c:   00000000 00000002 00000000 00000000                                         
0xb69b1b9c:   acc20440 00000000 b680615c 000000ec                                         
0xb69b1bac:   00000000 b6805800 acc28130 b69b1c80                                         
0xb69b1bbc:   b69b1cb8 b69b1c48 ac943970 ae2b1a38                                         
0xb69b1bcc:   b6805800 b69b1e64 b6f2a1f8 00000002                                         
0xb69b1bdc:   b69b1df8 b6f63000 b69b1df8 b6f59ce8                                         
                                                                                          
Instructions: (pc=0xb6f85ab8)                                                             
0xb6f85a98:   e1a02000 e5130454 e52d7004 e3a07f43                                         
0xb6f85aa8:   e5131458 e3500000 b2600000 ef000000                                         
0xb6f85ab8:   e3700a01 e49d7004 859f2010 82601000                                         
0xb6f85ac8:   83e00000 879f2002 87831002 e12fff1e                                         
                                                                                
Register to memory mapping:                                                               
                                                                                          
  r0  = 0x00000000                                                                        
[... cut ...]
                                                                                          
                                                                                          
                                                                                          
Stack: [0xb6964000,0xb69b4000],  sp=0xb69b1b6c,  free space=310k                          
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)           
C  [libpthread.so.0+0x10ab8]  raise+0x24
```

ocurr, 2016-01-22:
By save file do you mean the robot code?

JLLeitschuh, 2016-01-22:
No, the text of the `.grip` file you uploaded

ocurr, 2016-01-22:
Here you go:
```
<grip:Pipeline>
  <sources>
    <grip:Camera>
      <property name="address" value="http://10.9.73.20"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="resize">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1"/>
      <grip:Input step="0" socket="2">
        <value>0.5</value>
      </grip:Input>
      <grip:Input step="0" socket="3">
        <value>0.5</value>
      </grip:Input>
      <grip:Input step="0" socket="4">
        <value>INTER_LINEAR</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="RGB Threshold">
      <grip:Input step="1" socket="0"/>
      <grip:Input step="1" socket="1">
        <value>
          <double>185.74640287769782</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Input step="1" socket="2">
        <value>
          <double>0.0</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Input step="1" socket="3">
        <value>
          <double>206.38489208633092</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Output step="1" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Find Contours">
      <grip:Input step="2" socket="0"/>
      <grip:Input step="2" socket="1">
        <value>false</value>
      </grip:Input>
      <grip:Output step="2" socket="0" previewed="false"/>
    </grip:Step>
    <grip:Step name="Convex Hulls">
      <grip:Input step="3" socket="0"/>
      <grip:Output step="3" socket="0" previewed="false"/>
    </grip:Step>
    <grip:Step name="Filter Contours">
      <grip:Input step="4" socket="0"/>
      <grip:Input step="4" socket="1">
        <value>100.0</value>
      </grip:Input>
      <grip:Input step="4" socket="2">
        <value>0</value>
      </grip:Input>
      <grip:Input step="4" socket="3">
        <value>0</value>
      </grip:Input>
      <grip:Input step="4" socket="4">
        <value>1000</value>
      </grip:Input>
      <grip:Input step="4" socket="5">
        <value>0</value>
      </grip:Input>
      <grip:Input step="4" socket="6">
        <value>1000</value>
      </grip:Input>
      <grip:Output step="4" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Publish ContoursReport">
      <grip:Input step="5" socket="0"/>
      <grip:Input step="5" socket="1">
        <value>myContoursReport</value>
      </grip:Input>
      <grip:Input step="5" socket="2">
        <value>false</value>
      </grip:Input>
      <grip:Input step="5" socket="3">
        <value>false</value>
      </grip:Input>
      <grip:Input step="5" socket="4">
        <value>false</value>
      </grip:Input>
      <grip:Input step="5" socket="5">
        <value>true</value>
      </grip:Input>
      <grip:Input step="5" socket="6">
        <value>false</value>
      </grip:Input>
    </grip:Step>
  </steps>
  <connections>
    <grip:Connection>
      <grip:Output step="2" socket="0" previewed="false"/>
      <grip:Input step="3" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output source="0" socket="0" previewed="false"/>
      <grip:Input step="0" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="4" socket="0" previewed="true"/>
      <grip:Input step="5" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="3" socket="0" previewed="false"/>
      <grip:Input step="4" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="1" socket="0" previewed="true"/>
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
  </connections>
  <settings>
    <networkProtocol>NONE</networkProtocol>
    <teamNumber>0</teamNumber>
    <publishAddress></publishAddress>
    <deployAddress></deployAddress>
  </settings>
</grip:Pipeline>
```

JLLeitschuh, 2016-01-22:
Thank you! We will see what we can do.

closed 2017-01-08 by SamCarlberg