WPIRoboticsProjects/GRIP #413: Only show connectable sockets

JLLeitschuh, 2016-01-22:
SocketHint should have a boolean to hold whether it is connetable.

The `SocketHandleView` should check that value and be hidden based upon that boolean.

tom131313, 2016-01-26:
I had recognized 3 types of sockets and suggest 3 different appearances for inputs, outputs, and not really a socket at this time.
I like cues for usage (not knowing I had to connect the dots was the only criticism I had of my first 5 seconds with GRIP, I couldn't figure that out without help) and the male/female is always a good model.  Make the output plug mate with an input receptacle.
The 3rd (not a socket) should still be retained as an icon to denote that line is different than the other lines - like bullet points on a chart.  Thanks

AustinShalit, 2016-07-29:
@JLLeitschuh Do you think this is solved with the latest updates?

closed 2016-12-10 by AustinShalit

JLLeitschuh, 2016-12-10:
Yes