WPIRoboticsProjects/GRIP #420: VIDEOIO Error

stephenstockman, 2016-01-22:
Using a Microsoft lifecam hd3000 after the loaded no file step in deploy it states VIDEOIO ERROR:V4L/V4L2: VIDIOC_S_CROP
VIDEOIO ERROR:V4L: setting property #16 is not supported
Main start
Info:success

The grip file just has a webcam at 0; webdash board states the USB camera is at slot 0.

ThomasJClark, 2016-01-23:
V4L prints out lots of errors depending on what webcam you're using and what the driver supports.  Is anything not actually working?

stephenstockman, 2016-01-23:
Oh okay that's a relief, I'm still having issues with the c++ code in terms of it building but not deploying the Robot code to the bot(says success but Robot code drops). Ill attempt to get it working tmrr as I'm using a rio that has know deploy issues as our other two were being used. Thanks again for the quick replies.

ThomasJClark, 2016-01-23:
Cool.  Definitely open a new issue tomorrow if you can't get it to work

closed 2016-01-23 by ThomasJClark