WPIRoboticsProjects/GRIP #421: Documentation for GRIP execution on Drivers Station PC w/ Java

alec-hewitt, 2016-01-23:
Is this the same method as execution onboard the rio? If not, is there any info on how? Thanks.

ThomasJClark, 2016-01-23:
What specifically are you having a problem with?  Is GRIP not opening?  Are you trying to run in headless mode?

closed 2016-01-23 by alec-hewitt

KevinMackenzie, 2016-01-23:
I'm not the OP, but I have the same question.  I can open the GRIP ui and edit the pipeline, but how do I execute GRIP so it outputs the data to the network tables while running on the driver station computer?  When I try to just run "core-1.1.1-all.jar" on the driver station computer from command line I get an error

Exception in thread "main" java.lang.IllegalArgumentException: Unknown signal: HUP
        at sun.misc.Signal.<init>(Unknown Source)
        at edu.wpi.grip.core.Main.main(Main.java:37)

KevinMackenzie, 2016-01-23:
What did you do to solve this problem?

alec-hewitt, 2016-01-23:
Hi Kevin. Grip will automatically publish values to the network table if you are connected to the robots network and your camera is sending the feed. All you need to do is open the grip UI and play the camera.

Then check the network values with the eclipse outline viewer.

alec-hewitt, 2016-01-23:
Working on running grip from code, but grip opens up without my save file. As of now I can see the network values just by running grip. Make sure you have the publish contours report