WPIRoboticsProjects/GRIP #424: Run GRIP on driver station with network tables output

KevinMackenzie, 2016-01-23:
I have been looking around the internet, and have not been able to find any documentation for running grip on the diver station in such a way where it outputs to the network tables.  My question is how do I run GRIP on the driver station in 'deploy' mode.  I can edit the pipeline fine, but I cannot 'run' it.  Am I interpreting how the GRIP desktop ui works?  

P.S. I also saw someone else who seemed to have the same problem who closed the issue without posting how he did it.

ThomasJClark, 2016-01-23:
You don't deploy if you want to run on the driver station. Just set your team number in the project settings so it knows where to send NetworkTables values

KevinMackenzie, 2016-01-23:
ok, my issue must be somewhere else :/  Thanks for the clarification

closed 2016-01-23 by KevinMackenzie