WPIRoboticsProjects/GRIP #425: Roborio running out of memory

son34, 2016-01-23:
Our roboRIO is throwing this error: 

Loading Dependency Injection Framework
Feb 27, 2015 2:09:45 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Feb 27, 2015 2:09:45 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
platform: /Linux/arm/
Feb 27, 2015 2:09:50 AM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
Java HotSpot(TM) Embedded Client VM warning: INFO: os::commit_memory(0xb48c2000, 32768, 1) failed; error='Cannot allocate memory' (errno=12)
#
# There is insufficient memory for the Java Runtime Environment to continue.
# Native memory allocation (mmap) failed to map 32768 bytes for committing reserved memory.
# An error report file with more information is saved as:
# /home/lvuser/hs_err_pid1963.log
#
# Compiler replay data is saved as:
# /home/lvuser/replay_pid1963.log

Anyone else run into this problem? Or any ideas for fixes?

ThomasJClark, 2016-01-23:
Are you also running GRIP from your user program?  If so, there may be two instances  of GRIP running at the same time.  If this is the case, try commenting out the part of your program where you start GRIP, then restarting your RIO.

We might want to make it so any existing instances of GRIP are automatically killed when you deploy to avoid this

son34, 2016-01-23:
We are currently only deploying grip onto the roboRIO. We are essentially only trying to get it to run remotely, no actual code being used yet.

ThomasJClark, 2016-01-23:
Ok.  I guess try restarting, just in case there's another instance running for whatever reason.

If it's still a problem, can you post your pipeline?

son34, 2016-01-23:
Tries resetting the roboRIO, turned off and on the entire robot, as well as the computer (To be safe) but nothing seemed to fix it. Even checked the allocated memory on the roboRIO control panel in-browser. No problems there. 

And I'll post the pipeline in a couple secs

JLLeitschuh, 2016-01-23:
It's wherever you saved it. You can open the .grip file an a text editor. Just paste the contents here.

ThomasJClark, 2016-01-23:
Just take a screenshot of the pipeline

jakedp7, 2016-01-24:
@son34 and I's pipeline: 

[image]

```
<grip:Pipeline>
  <sources>
    <grip:ImageFile>
      <property name="path" value="C:\Users\Jake\Desktop\ultimate-ascent.jpeg"/>
    </grip:ImageFile>
    <grip:Camera>
      <property name="address" value="http://axis-camera.local"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="HSL Threshold">
    <grip:Step name="Blur">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1">
        <value>MEDIAN</value>
      </grip:Input>
      <grip:Input step="0" socket="2">
        <value>3.6036036036036037</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="HSL Threshold">
      <grip:Input step="1" socket="0"/>
      <grip:Input step="1" socket="1">
        <value>
          <double>87.41007194244605</double>
          <double>180.0</double>
          <double>69.60431654676258</double>
          <double>97.06484641638224</double>
        </value>
      </grip:Input>
      <grip:Input step="0" socket="2">
      <grip:Input step="1" socket="2">
        <value>
          <double>0.0</double>
          <double>116.95143884892086</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Input step="0" socket="3">
      <grip:Input step="1" socket="3">
        <value>
          <double>16.052158273381295</double>
          <double>75.67446043165468</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Output step="1" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Find Contours">
      <grip:Input step="1" socket="0"/>
      <grip:Input step="1" socket="1">
      <grip:Input step="2" socket="0"/>
      <grip:Input step="2" socket="1">
        <value>true</value>
      </grip:Input>
      <grip:Output step="1" socket="0" previewed="true"/>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Filter Contours">
      <grip:Input step="2" socket="0"/>
      <grip:Input step="2" socket="1">
        <value>100.0</value>
    <grip:Step name="Convex Hulls">
      <grip:Input step="3" socket="0"/>
      <grip:Output step="3" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Publish ContoursReport">
      <grip:Input step="4" socket="0"/>
      <grip:Input step="4" socket="1">
        <value>tapeTrackingCountours</value>
      </grip:Input>
      <grip:Input step="2" socket="2">
        <value>0.0</value>
      <grip:Input step="4" socket="2">
        <value>true</value>
      </grip:Input>
      <grip:Input step="2" socket="3">
        <value>0.0</value>
      <grip:Input step="4" socket="3">
        <value>true</value>
      </grip:Input>
      <grip:Input step="2" socket="4">
        <value>1000.0</value>
      <grip:Input step="4" socket="4">
        <value>true</value>
      </grip:Input>
      <grip:Input step="2" socket="5">
        <value>0.0</value>
      <grip:Input step="4" socket="5">
        <value>true</value>
      </grip:Input>
      <grip:Input step="2" socket="6">
        <value>1000.0</value>
      <grip:Input step="4" socket="6">
        <value>true</value>
      </grip:Input>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Convex Hulls">
      <grip:Input step="3" socket="0"/>
      <grip:Output step="3" socket="0" previewed="true"/>
    </grip:Step>
  </steps>
  <connections>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="1" socket="0" previewed="true"/>
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="2" socket="0" previewed="true"/>
      <grip:Input step="3" socket="0"/>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="3" socket="0" previewed="true"/>
      <grip:Input step="4" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output source="0" socket="0" previewed="true"/>
      <grip:Input step="0" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="2" socket="0" previewed="true"/>
      <grip:Input step="3" socket="0"/>
    </grip:Connection>
  </connections>
  <settings>
    <teamNumber>0</teamNumber>
    <teamNumber>2783</teamNumber>
    <publishAddress></publishAddress>
    <deployAddress></deployAddress>
    <deployAddress>roborio-2783-frc.local</deployAddress>
  </settings>
</grip:Pipeline>
```

ThomasJClark, 2016-01-24:
Try changing median blur to something else.  Median filters are really expensive in terms of CPU, although I'm not sure why that would lead to memory problems.

The text in the previous comment also doesn't seem to be a valid project file that GRIP would be able to create or open.  I'm not sure how it got corrupted, but maybe try recreating it

JLLeitschuh, 2016-01-24:
@ThomasJClark 
You'll need to remove the image source for you.

tom131313, 2016-01-25:
Here are a couple more examples of out of memory.  I may run the robot program 3 to 10 times without trouble then the next 1 to 4 times in a row get the out of memory.  Rebooting and restarting don't seem to have much effect in predicting when the error will happen.

There is only 1 java process running project.grip - the ps command display of processes appears okay.

Do you have a way of trying one or more of the suggestions in the error log to prevent this error?

ONE OF THE JAVA ERROR LOGS:

roboRIO-4237-FRC:/tmp$ cat hs_err_pid9636.log

 There is insufficient memory for the Java Runtime Environment to continue.
 Native memory allocation (mmap) failed to map 1048576 bytes for committing reserved memory.
 Possible reasons:
   The system is out of physical RAM or swap space
   In 32 bit mode, the process size limit was hit
 Possible solutions:
   Reduce memory load on the system
   Increase physical memory or swap space
   Check if swap backing store is full
   Use 64 bit Java on a 64 bit OS
   Decrease Java heap size (-Xmx/-Xms)
   Decrease number of Java threads
   Decrease Java thread stack sizes (-Xss)
   Set larger code cache with -XX:ReservedCodeCacheSize=
 This output file may be truncated or incomplete.

  Out of Memory Error (os_linux.cpp:2718), pid=9636, tid=3027186784

 JRE version: Java(TM) SE Embedded Runtime Environment (8.0_06-b23) (build 1.8.0_06-b23)
 Java VM: Java HotSpot(TM) Embedded Client VM (25.6-b23 mixed mode linux-arm )
 Core dump written. Default location: //core or core.9636 (max size 2048 kB). To ensure a full core dump, try "ulimit -c unlimited" before starting Java again

========================================
âž” Launching Â«'/home/lvuser/FRCUserProgram'Â»
[HAL] Starting task FRC_DriverStation...
[GRIP Test] in Constructor
[GRIP Test] in RobotInit
[GRIP Test] fork okay continuing with parent robot code
Default Disabled() method... Override me!
NT: ERROR: bind() failed: Address already in use (TCPAcceptor.cpp:102)
Loading Dependency Injection Framework
Jan 24, 2016 4:46:22 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 24, 2016 4:46:22 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
Java HotSpot(TM) Embedded Client VM warning: INFO: os::commit_memory(0xae53a000, 131072, 0) failed; error='Cannot allocate memory' (errno=12)

 There is insufficient memory for the Java Runtime Environment to continue.
 Native memory allocation (mmap) failed to map 131072 bytes for committing reserved memory.
 An error report file with more information is saved as:
 /tmp/hs_err_pid13039.log

=======

? Launching «'/home/lvuser/FRCUserProgram'»
[HAL] Starting task FRC_DriverStation...
GRIP Test in RobotInit
fork okay back in parent robot code
Default DisabledInit() method... Overload me!
Default DisabledPeriodic() method... Overload me!
NT: server: client CONNECTED: 169.254.248.235 port 51495
NT: server: client CONNECTED: 169.254.6.209 port 59054
NT: server: client CONNECTED: 169.254.6.209 port 59055
Loading Dependency Injection Framework
Jan 24, 2016 3:04:43 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 24, 2016 3:04:43 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
platform: /Linux/arm/
Jan 24, 2016 3:04:51 PM edu.wpi.grip.core.Main start
[... 189 lines cut ...]
      < grip:Output source="0" socket="0" previewed="false"/>
      < grip:Input step="0" socket="0"/>
    < /grip:Connection>
    < grip:Connection>
      < grip:Output step="0" socket="0" previewed="false"/>
      < grip:Input step="1" socket="0"/>
    < /grip:Connection>
    < grip:Connection>
      < grip:Output step="2" socket="0" previewed="false"/>
      < grip:Input step="3" socket="0"/>
    < /grip:Connection>
    < grip:Connection>
      < grip:Output step="1" socket="0" previewed="false"/>
      < grip:Input step="2" socket="0"/>
    < /grip:Connection>
  < /connections>
  < settings>
    < teamNumber>4237< /teamNumber>
    < publishAddress>roborio-4237-frc.local< /publishAddress>
    < deployAddress>roborio-4237-frc.local< /deployAddress>
    < deployDir>/home/lvuser< /deployDir>
    < deployUser>lvuser< /deployUser>
    < deployJavaHome>/usr/local/frc/JRE/< /deployJavaHome>
  < /settings>
< /grip:Pipeline>roboRIO-4237-FRC:~$

ThomasJClark, 2016-01-25:
So, the roboRIO *should* have enough memory to run this program.  After doing some profiling on my computer I have an educated guess about what might be happening.

Java usually tries to eat up as much memory as it's allowed to before it does garbage collection.  Most of the memory in GRIP is taken up by OpenCV objects, which are allocated from native code.  So, maybe the JRE doesn't know about them, so the heuristics for running the GC aren't working because they only take into account Java objects.

If this is right, we can fix it by either calling `System.gc()` periodically (which is usually frowned upon), or carefully calling `mat.release()` when necessary

tom131313, 2016-01-26:
hello

I spent a lot more time looking at GRIP running on the roboRIO,
restarting the roboRIO, rebooting the roboRIO, and deploying new robot
code from Eclipse to the roboRIO.

My testing is still not perfectly definitive and I hate to mislead you. 
I suspect that forcing garbage collection is not the solution to the
problem unless by periodically you mean as the program is initializing.

The out of memory is always near the beginning of a run while the
initialization messages are printing (within 10 seconds or much less).

I got a few out of memory errors during my latest experiments but only
one instance was I able to catch all the data I needed to know what might
have actually happened.

During deploy new code from Eclipse there is a killall and an execution
of FRCkill that should certainly clean up all old processes and then
start the newly deployed code.  Despite the seemingly double killing, the
old Java process running GRIP did NOT die.  It continued to live under
parent process ID = 1.  Yes there was then no end of errors thrown
because 2 GRIPs were running.  I can't imagine how that happened but it
did - I have the errors and "top" command output to prove 2 GRIPs at
once.

Is this my entire problem?  I don't know.  Maybe, maybe not.

What I really want to know is there enough processing power in the
roboRIO to run GRIP.  GRIP runs correctly much of the time and it's using
75% of the memory and 50% of the CPU with about 2 frames a second of
information.  Maybe I'd better stick with running GRIP on the fast driver
station PC.  An advantage is we could then see what the images are that
are being processed (I hate flying blind on the roboRIO).  I tried that
last night with an IP Axis M1011 and the camera image froze in GRIP after
running a minute although the camera itself continued to stream images to
a browser connection that I started while GRIP was frozen.  So that GRIP
didn't work either and I clicked on send error report and an issue was
created (sorry, it was probably a useless print of the time out error).

I think GRIP is an extremely good concept for the FRC team - reasonably
fast (maybe, on the right computer) and very easy to use compared to all
other computer vision programs we have available.  The kids only get to
spend a few hours developing computer vision and GRIP seems to be the
tool of choice if we can get it running well.

Thanks for your help.  I greatly appreciate your effort for developing
GRIP.

Rick Thomas
========

Richard K. Thomas

tom131313, 2016-01-26:
One more conjecture that might be related to out of memory -- timing
conflicts in handling log/lck files?

I'm starting to suspect there is a timing and resource conflict between
terminating the previous instance of GRIP and creating the new instance. 
Maybe stuff happens in the new run on the roboRIO before the old run is
cleaned up.

Sometimes "cannot allocate memory" error is accompanied by "No space left
on device" error.

There is no hint for me about what device that would be.  There is plenty
of space on the "disk" drive (190.2M used, 191.4M available from the df
command).

The only thing I see strange about files is the use of the GRIP.log
(.sequence#) and GRIP.log (.sequence#).lck.

Apparently when the driver station restarts the robot program, the new
instance of Java GRIP reuses the previous GRIP log files.

When the roboRIO is rebooted the next available sequence number (starting
from the beginning and not necessarily the last used number) is used.

Strangely an occasional log or lck file will disappear from the file list
(ls -l).

Further strangely, the date/time stamp of files change - especially the
lck files and recently created files (a couple of minutes ago) will
reappear on restart as having 13 day old dates and times.

Maybe these observations will help; hope that don't confuse.

Rick

tom131313, 2016-01-26:
In another essentially identical issue Mr Clark suggested checking for an extra java process before starting another java grip.  I used his code to check and kill the old java before starting another.  It seems to have cured (along with some Waiting) all my ills but one.  There is still a port resource that has a conflict.  Here's the error(s) I'm still left with and my code that runs except for this error after deploying from Eclipse:

I do not know what process is going after the port in question so I don't know what to do about it.

Running GRIP on the roboRIO:

Good message often appears IMMEDIATELY after RobotInit starting message prints from a reboot:
NT: server: client CONNECTED: 169.254.6.209 port 64809

Bad message often appears in its place after a deploy from Eclipse:
NT: ERROR: bind() failed: Address already in use (TCPAcceptor.cpp:102)

No message appears usually after a restart robot code (driver station function)

And then there are more of these bad messages coming out farther below after the Eclipse deploy fails with the above message:
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to roborio-4237-frc.local port 1735 error 111 - Connection refused

=========
[HAL] Starting task FRC_DriverStation...
[GRIP Test] in Constructor
[GRIP Test] Compiled file:..\src\Robot.cpp, date:Jan 26 2016, time:14:58:34, function:Robot, object address:0xbf420, function:Robot::Robot(), printed at line:19
[GRIP Test] in RobotInit
NT: ERROR: bind() failed: Address already in use (TCPAcceptor.cpp:102)
7724
killed java with kill 7724
[GRIP Test] fork okay, continuing with robot code.
[GRIP Test] Parent PID 7888; this robot PID 7891; JAVA GRIP in child PID=7946
Loading Dependency Injection Framework
Jan 26, 2016 8:16:35 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 26, 2016 8:16:35 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
platform: /Linux/arm/
Jan 26, 2016 8:16:39 PM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
VIDEOIO ERROR: V4L/V4L2: VIDIOC_S_CROP
VIDEOIO ERROR: V4L: setting property #16 is not supported
Corrupt JPEG data: 2 extraneous bytes before marker 0xd4
Corrupt JPEG data: 1 extraneous bytes before marker 0xd3
Jan 26, 2016 8:16:43 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Jan 26, 2016 8:16:43 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
many lines deleted
Jan 26, 2016 8:16:45 PM edu.wpi.grip.core.Main onExceptionClearedEvent
INFO: Exception Cleared Event
Jan 26, 2016 8:16:45 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to  port 1735 error 111 - Connection refused
Jan 26, 2016 8:16:46 PM edu.wpi.grip.core.Main onExceptionClearedEvent
INFO: Exception Cleared Event
Jan 26, 2016 8:16:46 PM edu.wpi.grip.core.Main onExceptionClearedEvent
INFO: Exception Cleared Event
Corrupt JPEG data: 1 extraneous bytes before marker 0xd2
Jan 26, 2016 8:16:46 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to  port 1735 error 111 - Connection refused
Jan 26, 2016 8:16:46 PM edu.wpi.grip.core.Main start
INFO: SUCCESS! The project is running in headless mode!
Corrupt JPEG data: 2 extraneous bytes before marker 0xd0
Jan 26, 2016 8:16:47 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to roborio-4237-frc.local port 1735 error 111 - Connection refused
Corrupt JPEG data: 4 extraneous bytes before marker 0xd7
Jan 26, 2016 8:16:47 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to roborio-4237-frc.local port 1735 error 111 - Connection refused
Corrupt JPEG data: 1 extraneous bytes before marker 0xd6
Jan 26, 2016 8:16:48 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to roborio-4237-frc.local port 1735 error 111 - Connection refused
Corrupt JPEG data: 1 extraneous bytes before marker 0xd6
Jan 26, 2016 8:16:48 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$8
SEVERE: NetworkTables: TCPConnector.cpp:157 select() to roborio-4237-frc.local port 1735 error 111 - Connection refused
Default Disabled() method... Override me!
[... 101 lines cut ...]
			Wait(.1);
			continue;
			}
			prev_blob = first_blob;

			for (auto blob : blobsx) {
				std::cout << "[GRIP Test] Got blob with x=" << blob << std::endl;
			}

			auto blobsy = grip->GetNumberArray("myBlobsReport/y", init /*llvm::ArrayRef<double>()*/);
			for (auto blob : blobsy) {
				std::cout << "[GRIP Test] Got blob with y=" << blob << std::endl;
			}

			auto blobssize = grip->GetNumberArray("myBlobsReport/size", init /*llvm::ArrayRef<double>()*/);
			for (auto blob : blobssize) {
				std::cout << "[GRIP Test] Got blob with size=" << blob << std::endl;
			}
			Wait(.1);
		}
		SmartDashboard::PutNumber("[GRIP Test line]", __LINE__);
	}
};

START_ROBOT_CLASS(Robot)

JLLeitschuh, 2016-01-26:
In order to make your messages more readable please use markdown to format your messages
https://guides.github.com/features/mastering-markdown/

If you edit your messages above you will see I have reformatted your messages so they can be read easier.
Thanks.

closed 2016-02-01 by JLLeitschuh

mdmedoff, 2016-02-20:
tom131313.  You stated "In another essentially identical issue Mr Clark suggested checking for an extra java process before starting another java grip. I used his code to check and kill the old java before starting another."  I could not find this issue.  Can you either post this code or post a link to this code?  Thanks.

tom131313, 2016-02-20:
@mdmedoff  I think you are asking where is Mr. Clark's code.  Not sure anymore and I think he is now publishing a LInux script to start GRIP on the roboRIO that includes killing previous processes probably with a signal 9 included but not sure.  I'll post here my last version of the GRIP troubleshooting/testing that I did.  Sorry - it's really hacked up and includes multiple kills and other junk.  Look at the uncommented stuff for what I thought was best.  I'm not much of a LInux person anymore and now remember that "killall java" may be much better than my looping through all PIDs that should be killed.  Previous kill examples only killed 1 PID and that was not sufficient for my multiple processes running situation (which I have no idea how it happened except there may be significant delays between requesting a kill and it actually happening and also children might no be automatically killed when the parent is).  If you want something other than what I pasted below, please let me know.  We pursued multiple paths and did get running GRIP on roboRIO, GRIP on Windows driver station, and GRIP on Raspberry Pi.  Also, used roboRealm on Windows driverstation and decided to go with roboRealm running on a Kangaroo Windows on-board the robot and that is what the student team members wanted to do and now we are trying to keep our gears from being chewed up on our shooter (vision is slightly lower priority until Wednesday).  Good luck with your competition.

```C++
#include <unistd.h> // fork, execv, getppid, getpid
#include <stdio.h>
#include "WPILib.h"

class Robot: public SampleRobot
{
	std::shared_ptr<NetworkTable> grip;

    const char *JAVA = "/usr/local/frc/JRE/bin/java";

    char *GRIP_ARGS[5] = { "java", "-jar", "/home/lvuser/grip.jar", "/home/lvuser/project.grip", NULL };

public:

	Robot():grip(NetworkTable::GetTable("GRIP")){
		std::cout << "[GRIP Test] in Constructor" << std::endl;
//		SmartDashboard::PutNumber("[GRIP Test line]", __LINE__);
		printf("[GRIP Test] Compiled file:%s, date:%s, time:%s, function:%s, object address:%p, function:%s, printed at line:%d\n",
				__FILE__, __DATE__, __TIME__, __FUNCTION__, this, __PRETTY_FUNCTION__, __LINE__);
	}

    void RobotInit() override {
    	std::cout << "[GRIP Test] in RobotInit" << std::endl;
		//SmartDashboard::PutNumber("[GRIP Test line]", __LINE__);

		Wait(1.); // wait for previous instances of GRIP to end

		/* kill any java processes still running */
		char PIDjava[100]; // process id list
		char killstring[100]; // kill string for a process

		FILE *cmd = popen("pidof java", "r");

		while (fscanf (cmd, "%s", PIDjava) > 0)
			{
			strcpy(killstring, "kill ");
			strcat(killstring, PIDjava);
			std::cout << "killing java with " << PIDjava << std::endl;
			system(killstring);
			}

		pclose(cmd);

        /* Run GRIP in a new process */
        int retval=fork();
        Wait(3.);
        if ( retval == 0) {
        	// in the child process so start Java with the GRIP program
            if (execv(JAVA, GRIP_ARGS) == -1) {
                perror("[GRIP Test] Error running GRIP in new process");
            }
        }
        // in the parent process
        else if (retval == -1) printf("[GRIP Test] fork error returned to parent error %d\n", retval);
        else printf("[GRIP Test] fork okay, continuing with robot code.\n[GRIP Test] Parent PID %d; this robot PID %d; JAVA GRIP in child PID=%d\n", getppid(), getpid(), retval);
        Wait(5.); // wait for GRIP to initialize
    }

//		char line[20];
//		FILE *cmd = popen("pidof java", "r");
//		fgets(line, 20, cmd);
//		pid_t pid = strtoul(line, NULL, 10);
//		pclose(cmd);
//		std::cout << pid << std::endl;
//
//		if (pid > 0)
//		{char killstring[20]="kill 0";
[... 81 lines cut ...]
			if (Targetarea[1] != init[1]) std::cout << "area" << std::endl;
			if (Targetsolidity[1] != init[1]) std::cout << "solidity" << std::endl;
			std::cout << TargetcenterX[1]  << " " << TargetcenterY[1]  << " " << Targetwidth[1]  << " " << Targetheight[1]  << " " << Targetarea[1]  << " " << Targetsolidity[1]  << std::endl;
			Wait(.1);
		}
		std::cout << "[GRIP Test] leaving OperatorControl" << std::endl;
//
//		SmartDashboard::PutNumber("[GRIP Test line]", __LINE__);
	}
};

int main() {
    if (!HALInitialize()) {
    	std::cerr << "FATAL ERROR: HAL could not be initialized" << std::endl;
      return -1;
    }
    HALReport(HALUsageReporting::kResourceType_Language,
              HALUsageReporting::kLanguage_CPlusPlus);
    Robot *robot = new Robot();
    RobotBase::robotSetup(robot);
    std::cout << "[GRIP Test] leaving main" << std::endl;
    return 0;
  }
//START_ROBOT_CLASS(Robot)
```

PR #454 (2016-02-01): Add JVM args and more options to the deploy dialog
JVM arguments can be specified, and some reasonable defaults are included for new projects (and old projects created before this change).  These settings should help reduce OOM errors, and help us debug new ones.

Running from a robot program is now also easier, since a new wrapper script is included to facilitate the JVM options.

```c++
void RobotInit() override {
    system("/home/lvuser/grip &");
}
```

In addition, the resulting file name of the deployed project can be changed from the default of "project.grip".  This lets you upload mult [... cut]