WPIRoboticsProjects/GRIP #429: GRIP run inside RoboRIO with Microsoft HD-3000 init fail

guoxiangyang, 2016-01-24:
run grip in console:
```
/usr/local/frc/JRE//bin/java -jar /home/lvuser/grip.jar /home/lvuser/project.grip
```

Error message;
```
roboRIO-5522-FRC:~$ /usr/local/frc/JRE//bin/java -jar /home/lvuser/grip.jar /home/lvuser/project.grip
Loading Dependency Injection Framework
Jan 24, 2016 5:25:49 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 24, 2016 5:25:49 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
platform: /Linux/arm/
Jan 24, 2016 5:25:59 AM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: setting property #16 is not supported
VIDIOC_STREAMON: Bad file descriptor
VIDIOC_STREAMON: Bad file descriptor
Jan 24, 2016 5:26:05 AM edu.wpi.grip.core.Source initializeSafely
WARNING: Failed to initialize CameraSource
java.io.IOException: A problem occurred trying to start the frame grabber for Webcam 0
	at edu.wpi.grip.core.sources.CameraSource.start(CameraSource.java:191)
	at edu.wpi.grip.core.sources.CameraSource.initialize(CameraSource.java:174)
	at edu.wpi.grip.core.Source.initializeSafely(Source.java:98)
	at edu.wpi.grip.core.serialization.SourceConverter.unmarshal(SourceConverter.java:60)

```
Project.grip

```
<grip:Pipeline>
  <sources>
    <grip:Camera>
      <property name="deviceNumber" value="0"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="CV resize">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1"/>
      <grip:Input step="0" socket="2">
        <value>0.25</value>
      </grip:Input>
      <grip:Input step="0" socket="3">
        <value>0.25</value>
      </grip:Input>
      <grip:Input step="0" socket="4">
        <value>INTER_LINEAR</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="false"/>
    </grip:Step>
    ......
```

guoxiangyang, 2016-01-24:
[image]

guoxiangyang, 2016-01-24:
Resolved.

GRIP conflict with Simple Vision demo.

closed 2016-01-24 by guoxiangyang

Ubspy, 2016-12-09:
@guoxiangyang How'd you fix it? I'm having the same issue.