WPIRoboticsProjects/GRIP #430: Add maintainer address to package

robobenklein, 2016-01-24:
[image]

Although not absolutely needed, it would make automated and script installs much easier. Right now some install scripts will simply fail when the package does not meet quality standards.

Ubuntu Software Center at least has an option to override.

PaulaRudy, 2016-01-25:
Thanks for your submission!  This is a known problem right now, but it's always helpful to have an issue open for us to keep track of.

ThomasJClark, 2016-01-25:
@paularudy, can you do this since you're on Ubuntu?

JLLeitschuh, 2016-01-25:
Should probably just be a setting in out build.gradle
There should be an example somewhere in here: https://bitbucket.org/shemnon/javafx-gradle

PaulaRudy, 2016-01-26:
@JLLeitschuh  the bitbucket address gives me nothing. Apparently he's stopped maintaining it :(

JLLeitschuh, 2016-01-26:
Thats what we are using for our build.
https://bitbucket.org/shemnon/javafx-gradle/src/4695a6d1ed9e4b9248c93a877ee8374c16b9105b/samples/?at=master
There are a bunch of examples you can find in these subfolders.

closed 2016-01-27 by JLLeitschuh