WPIRoboticsProjects/GRIP #432: ContoursReport publishing incorrect NetworkTable keys

tom131313, 2016-01-25:
ContoursReport publishing user selected values "area" and "height" creates NetworkTables keys "height" and "centerY" (and once I saw "solidity" and "height".

Fields are passed "positionally" and have different names in the User Interface and the NetworkTables.  I didn't check for any other instances of this type of error.

Portion of .grip file deployed to the roboRIO:
   < grip:Step name="Publish ContoursReport">
   < /grip:Input>
      < grip:Input step="3" socket="5">
        < value>true</value>
      < /grip:Input>
      < grip:Input step="3" socket="6">
        < value>true</value>

ThomasJClark, 2016-01-27:
Can you post  a screenshot of your pipeline and outlineviewer?

tom131313, 2016-01-28:
I'm sorry for the delay for providing more information.

Most of our team's PCs are 32 bit and include a small ClassMate (and all
team PCs are Windows 7).  I was happy GRIP ran on the ClassMate.  When
you distributed GRIP v1.1.1 temporarily without 32 bit support I was able
to borrow our 64 bit machine to continue evaluation of GRIP but now the
64 bit machine is being used elsewhere.

When I'm able to get the 64 bit machine again, I'll respond.  Or maybe
you will distribute the 32 bit version before I am able to use the 64 bit
machine.  Thanks.

tom131313, 2016-01-29:
Steps to test error:

Notice Contour Report has "height" and "area."
[image]

Deploy the GRIP project on the driver station PC to the roboRIO.

(Using PuTTY to the roboRIO I see a java process is started on the roboRIO.)

Stop the GRIP driver Station deploy and close the deploy window.  (Doesn't seem to matter when the deploy is stopped, actually.)

Deploy my C++ robot program that uses GRIP on the roboRIO from Eclipse.
(A new java process is created.)

Open the Network Table Viewer in Eclipse on the driver station PC (client mode).

[image]

Notice that "MyTargetTable" has "height" and "solidity" even though the GRIP project has "height" and "area."

__CURIOUSLY!!!:__
If I click on the driver station PC GRIP window (gets focus) and then click on the Network Table Viewer window (gets focus), then the "area" does appear in place of "solidity" for about 1/2 second in the Network Table Viewer and then returns to "solidity."  The "area" appears briefly despite the GRIP deploy not running (stopped and closed the deploy window).  Closing the GRIP program on the driver station PC doesn't change the Network Table Viewer - still shows "height" and "solidity."

cartycrabber, 2016-01-29:
Confirming this report, it seems that the check boxes are linked to different values than they display. Unchecking "Publish Solidity" results in center x not being published while solidity is. Unchecking "Publish Center X" results in width not being published while center x is, and so on.

JLLeitschuh, 2016-01-29:
@ThomasJClark Can you look into this one. This sounds like a blocking issue.

ThomasJClark, 2016-02-03:
The publish operations get the list of fields you can publish using Java reflection.  This is super helpful for maintaining the code, since we can can just have a "Publish *x*" step for any given *x* without writing a bunch of boilerplate code.  Unfortunately, it turns out that the order that Java gives us the list of fields is actually undefined behavior, so it can return them in different orders on different JVM versions.

What's happening here is that your desktop Java install gives `area` and `height` as the 4th and 5th fields of `ContourReport`, so you've checked the 4th and 5th checkboxes.  However, the roboRIO's JVM returns the fields in such an order that `height` and `solidity` are the 4th and 5th ones, so it publishes those.

The best solution is likely to make it so the `ContoursReport` class (and other similar classes) can explicitly define an index for each publishable field. I'll begin work on that.

PR #468 (2016-02-04): Fix NTPublishOperation problems
This fixes two problems:
- Publishable fields now much have a "weight".  This determines the
  order that the checkboxes are in, preventing issues with different JVM
  versions returning methods in different orders.
- If a publishable value has no (sub) key, it's put directly in the key
  for the entire step.  This was sort of the behavior before, but
  previously NetworkTables automatically put a trailing slash at the
  end.  OutlineViewer still shows it the same, but robot programs did
  not work unless they included this trailing slash.

Closes #432
commit 344b4533 (2016-02-04): Fix NTPublishOperation problems

This fixes two problems:
- Publishable fields now much have a "weight".  This determines the
  order that the checkboxes are in, preventing issues with different JVM
  versions returning methods in different orders.
- If a publishable value has no (sub) key, it's put directly in the key
  for the entire step.  This was sort of the behavior before, but
  previously NetworkTables automatically put a trailing slash at the
  end.  OutlineViewer still shows it the same, but robot programs did
  not work unless they included this trailing slash.

Closes #432
closed 2016-02-04 by ThomasJClark