WPIRoboticsProjects/GRIP #435: Problem deploying GRIP

FRC3314, 2016-01-25:
[image]

After having updated GRIP, we are unable to deploy GRIP to the roboRIO, despite it having worked before. Following the path, we clearly have the .jar file, but GRIP says that it is "not a regular file or directory." We have no idea why it isn't working.

Any help would be appreciated. Thank you.

ThomasJClark, 2016-01-26:
It probably has something to do with the path having a space in it.  If possible, can you try making a user account without a space and trying again?

This should be easy for us to fix, but in the meantime I think that will work.

FRC3314, 2016-01-26:
This did solve the issue. Thank you for the timely response.

ThomasJClark, 2016-01-26:
Glad you got it to work, but I'm gonna keep this open until we fix the underlying problem

PR #437 (2016-01-26): Fix lack of URL decoding when deploying
This was causing problems when the full path of the local GRIP jar has spaces in it.

Closes #435
commit 02c4a210 (2016-01-26): Fix lack of URL decoding when deploying

This was causing problems when the full path of the local GRIP jar has
spaces in it.

Closes #435
closed 2016-01-26 by JLLeitschuh