WPIRoboticsProjects/GRIP #441: Add Advanced Filter Contours capabilities.

ThadHouse, 2016-01-26:
This could easily be a 2nd Filter Contours operation command, but it would be nice to have the following filters.

* Vertex Count (Allows easy filtering of rectangles)
* Force Convex (Allows easy filtering of odd shapes)
* Ratio filter (Check height to width ratio, and filter if out of range)
* Parallelism filter (Check how many lines are vertical and horizontal)

With these 4 added, it would make it much easier to force only the target to be seen and filter in GRIP, when we have more information.

JLLeitschuh, 2016-01-26:
We welcome the addition of more operations.
Perhaps we should create a wiki topic about how to write/add operation PR's

AustinShalit, 2016-07-29:
This issue was solved with #531.  If you think this is not the case, please reopen this issue.

closed 2016-07-29 by AustinShalit