WPIRoboticsProjects/GRIP #445: Getting more done in GitHub with ZenHub

JLLeitschuh, 2016-01-27:
Hola! @JLLeitschuh has created a [ZenHub](http://www.zenhub.io) account for the **WPIRoboticsProjects** organization. ZenHub is the leading team collaboration and project management solution built for GitHub.

----

#### How do I use ZenHub?

To get set up with ZenHub, all you have to do is **[download the browser extension](https://www.zenhub.io?utm_source=ZHOnboarding)** and log in with your GitHub account. Once you do, you’ll get access to ZenHub’s complete feature-set immediately.

#### What can ZenHub do?

ZenHub adds a series of enhancements directly inside the GitHub UI:

- Real-time, customizable task boards for GitHub issues;
- Burndown charts, estimates, and velocity tracking based on GitHub Milestones;
- Personal to-do lists and task prioritization;
- “+1” button for GitHub issues and comments;
- Drag-and-drop file sharing;
- Time-saving shortcuts like a quick repo switcher.

### [Add ZenHub to GitHub](https://www.zenhub.io?utm_source=ZHOnboarding)

_Still curious? See [more ZenHub features](https://www.zenhub.io/features?utm_source=ZHOnboarding) or read [user reviews](https://chrome.google.com/webstore/detail/zenhub-for-github/ogcgkffhplmphkaahpmffcafajaocjbd/reviews). This issue was written by your friendly ZenHub bot, posted by request from @JLLeitschuh._

[image]

closed 2016-01-27 by ThomasJClark

JLLeitschuh, 2016-01-27:
You should take a look at it. Its really cool.

ThomasJClark, 2016-01-27:
ok