WPIRoboticsProjects/GRIP #448: Opening file in GRIP spams console with UI errors

rlee287, 2016-01-28:
When loading a file, this gets spammed into the console. The part that says 'Button.delete' is different each time, so it must be referring to different GUI elements.
```Jan 27, 2016 4:01:29 PM javafx.scene.CssStyleHelper calculateValue
WARNING: Could not resolve '-fx-dark-color' while resolving lookups for '-fx-text-fill' from rule 'Button.delete' in stylesheet file:/C:/Users/Ryan/Documents/GitHub/Robotics/GRIP/ui/build/resources/main/edu/wpi/grip/ui/GRIP.css```

JLLeitschuh, 2016-01-28:
This isn't anything that breaks the UI but this is something we could do to clean up our console spam. 
Thanks!
Line number creating the problem.
https://github.com/WPIRoboticsProjects/GRIP/blob/master/ui/src/main/resources/edu/wpi/grip/ui/GRIP.css#L112

JLLeitschuh, 2016-02-13:
This hasn't been resolved.
The CSS is still wrong.