WPIRoboticsProjects/GRIP #459: Publishing network tables from Kangaroo to RoboRIO

FRC3314, 2016-01-30:
[image]
How would you go about doing this? We've set the Kangaroo to publish the network tables to the RIO, but it doesn't seem to be receiving them. When we check the network tables, there's no data. We're connected to the RIO by wifi, if that matters.

ThomasJClark, 2016-01-31:
Are you checking the values using a robot program, or using OutlineViewer?  If it's OutlineViewer, are you in server or client mode?

Are you sure the kangaroo can connect to the rio?  Can you ping it?

FRC3314, 2016-01-31:
We're trying to get the Kangaroo to send the values to the rio, and then we're viewing them on a laptop using OutlineViewer in client mode. We can see the values when we run GRIP off the laptop, but not when we run it off the Kangaroo. 

The Kangaroo seems to be connecting to the rio, as we can ping the rio from the Kangaroo.

bradamiller, 2016-01-31:
I've been running GRIP on a kangaroo connected to RoboRIO and it's been working for me. Can you make sure that you have the networking set up correctly on the kangaroo. Maybe you can verify that you can ping the roboRIO from the kangaroo when it's connected to a monitor. In fact if you're running wirelessly then you can have the kangaroo off the robot and keep it next to a monitor communicating to the robot access point. Then you can keep a keyboard and mouse connected to it for easier testing.

FRC3314, 2016-01-31:
[image]

We have successfully pinged the Roborio from the kangaroo. Our current setup is the kangaroo connected wirelessly to the robot with the monitor, keyboard, and mouse connected to the kangaroo.

ThomasJClark, 2016-01-31:
That's not the same address you have set as your "NetworkTables server address".

Try changing it to 10.33.14.58, and try pinging roborio-3314-frc.local.

FRC3314, 2016-01-31:
[image]
It seems we are unable to find the rio. We changed the network tables server address to 10.33.14.58

bradamiller, 2016-01-31:
You could always put the actual robot IP address into the Kangaroo instead of relying on the team number and the mDNS lookup, then the network tables client should connect.

ThomasJClark, 2016-01-31:
Are you sure the roboRIO's address is 10.33.14.58?  Usually it's something like 10.33.14.2.

FRC3314, 2016-01-31:
We re-changed the networktables server address to 10.33.14.58 and it seems to be working now. We don't know why it began it work now seeing as we have already done this, but it is now working. Thanks

closed 2016-01-31 by ThomasJClark