WPIRoboticsProjects/GRIP #460: Deploy TimeoutException

CubeMaster007, 2016-01-31:
Deploy a pipeline to a roborio over a wireless connection.
Error:

An event subscriber threw an exception
## Stack Trace:
```java
java.util.concurrent.TimeoutException: Unable to terminate video feed from IP Camera 10.90.1.20
	at edu.wpi.grip.core.sources.CameraSource.stop(CameraSource.java:279)
	at edu.wpi.grip.core.sources.CameraSource.onStopPipeline(CameraSource.java:339)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.ui.DeployController.deploy(DeployController.java:157)
	at edu.wpi.grip.ui.DeployController.lambda$onDeploy$8(DeployController.java:108)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.1.1
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 7
os.version | 6.1
os.arch | amd64

ThomasJClark, 2016-01-31:
I think this is fixed by @JLLeitschuh's camera thing

CubeMaster007, 2016-01-31:
I tried deploying a second time with the same settings and it worked. I'm not sure why it failed in the first place. Thanks though

closed 2016-01-31 by ThomasJClark

JLLeitschuh, 2016-01-31:
This will be solved by #456

commit dbc156d1 (2016-02-05): Fixes IP Cameras Hanging

 - The IPCameraFrameGrabber was stuck in an infinte loop
waiting for the rest of the image to arrive.
This was fixed by using a DataInputStream instead
 - An unexpected end of the stream would cause a null
pointer exception. Now throws an EOFException. #390

Closes #390
Closes #427
Closes #404
Closes #387
Closes #436
Closes #460
Related https://github.com/bytedeco/javacv/issues/322