WPIRoboticsProjects/GRIP #461: Error when building(will update title upon more debugging)

rlee287, 2016-01-31:
Current version (34fd6bee78391fb049e517ea0e9a8f1d101303df) broken:
```
[sts] -----------------------------------------------------
[sts] Starting Gradle build for the following tasks: 
[sts]      :ui:run
[sts] -----------------------------------------------------
:buildSrc:compileJava UP-TO-DATE
:buildSrc:extractSources UP-TO-DATE
:buildSrc:compileGroovy UP-TO-DATE
:buildSrc:processResources UP-TO-DATE
:buildSrc:classes UP-TO-DATE
:buildSrc:jar UP-TO-DATE
:buildSrc:assemble UP-TO-DATE
:buildSrc:compileTestJava UP-TO-DATE
:buildSrc:compileTestGroovy UP-TO-DATE
:buildSrc:processTestResources UP-TO-DATE
:buildSrc:testClasses UP-TO-DATE
:buildSrc:test UP-TO-DATE
:buildSrc:check UP-TO-DATE
:buildSrc:build UP-TO-DATE

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred evaluating root project 'GRIP'.
> A problem occurred starting process 'command 'git''

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 3.495 secs
[sts] Build failed
org.gradle.tooling.BuildException: Could not execute build using Gradle distribution 'https://services.gradle.org/distributions/gradle-2.10-bin.zip'.
	at org.gradle.tooling.internal.consumer.ResultHandlerAdapter.onFailure(ResultHandlerAdapter.java:57)
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:57)
	at org.gradle.internal.concurrent.DefaultExecutorFactory$StoppableExecutorImpl$1.run(DefaultExecutorFactory.java:64)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
	at org.gradle.tooling.internal.consumer.BlockingResultHandler.getResult(BlockingResultHandler.java:46)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher.run(DefaultBuildLauncher.java:71)
	at org.springsource.ide.eclipse.gradle.core.TaskUtil.execute(TaskUtil.java:117)
	at org.springsource.ide.eclipse.gradle.core.launch.GradleProcess$1.doit(GradleProcess.java:92)
	at org.springsource.ide.eclipse.gradle.core.util.GradleRunnable$1.run(GradleRunnable.java:53)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
Caused by: org.gradle.internal.exceptions.LocationAwareException: A problem occurred evaluating root project 'GRIP'.
	at org.gradle.initialization.DefaultExceptionAnalyser.transform(DefaultExceptionAnalyser.java:74)
	at org.gradle.initialization.MultipleBuildFailuresExceptionAnalyser.transform(MultipleBuildFailuresExceptionAnalyser.java:47)
	at org.gradle.initialization.StackTraceSanitizingExceptionAnalyser.transform(StackTraceSanitizingExceptionAnalyser.java:30)
	at org.gradle.initialization.DefaultGradleLauncher$1.create(DefaultGradleLauncher.java:101)
	at org.gradle.initialization.DefaultGradleLauncher$1.create(DefaultGradleLauncher.java:93)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:90)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:62)
	at org.gradle.initialization.DefaultGradleLauncher.doBuild(DefaultGradleLauncher.java:93)
	at org.gradle.initialization.DefaultGradleLauncher.run(DefaultGradleLauncher.java:82)
	at org.gradle.launcher.exec.InProcessBuildActionExecuter$DefaultBuildController.run(InProcessBuildActionExecuter.java:94)
	at org.gradle.tooling.internal.provider.runner.BuildModelActionRunner.run(BuildModelActionRunner.java:46)
	at org.gradle.launcher.exec.ChainingBuildActionRunner.run(ChainingBuildActionRunner.java:35)
	at org.gradle.tooling.internal.provider.runner.SubscribableBuildActionRunner.run(SubscribableBuildActionRunner.java:58)
	at org.gradle.launcher.exec.ChainingBuildActionRunner.run(ChainingBuildActionRunner.java:35)
	at org.gradle.launcher.exec.InProcessBuildActionExecuter.execute(InProcessBuildActionExecuter.java:43)
	at org.gradle.launcher.exec.InProcessBuildActionExecuter.execute(InProcessBuildActionExecuter.java:28)
	at org.gradle.launcher.exec.ContinuousBuildActionExecuter.execute(ContinuousBuildActionExecuter.java:78)
	at org.gradle.launcher.exec.ContinuousBuildActionExecuter.execute(ContinuousBuildActionExecuter.java:48)
	at org.gradle.launcher.daemon.server.exec.ExecuteBuild.doBuild(ExecuteBuild.java:52)
	at org.gradle.launcher.daemon.server.exec.BuildCommandOnly.execute(BuildCommandOnly.java:36)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.WatchForDisconnection.execute(WatchForDisconnection.java:37)
[... 41 lines cut ...]
	at org.gradle.initialization.DefaultGradleLauncher.doBuildStages(DefaultGradleLauncher.java:122)
	at org.gradle.initialization.DefaultGradleLauncher.access$200(DefaultGradleLauncher.java:32)
	at org.gradle.initialization.DefaultGradleLauncher$1.create(DefaultGradleLauncher.java:99)
	... 42 more
Caused by: org.gradle.process.internal.ExecException: A problem occurred starting process 'command 'git''
	at org.gradle.process.internal.DefaultExecHandle.setEndStateInfo(DefaultExecHandle.java:197)
	at org.gradle.process.internal.DefaultExecHandle.failed(DefaultExecHandle.java:327)
	at org.gradle.process.internal.ExecHandleRunner.run(ExecHandleRunner.java:86)
	... 2 more
Caused by: net.rubygrapefruit.platform.NativeException: Could not start 'git'
	at net.rubygrapefruit.platform.internal.DefaultProcessLauncher.start(DefaultProcessLauncher.java:27)
	at net.rubygrapefruit.platform.internal.WindowsProcessLauncher.start(WindowsProcessLauncher.java:22)
	at net.rubygrapefruit.platform.internal.WrapperProcessLauncher.start(WrapperProcessLauncher.java:36)
	at org.gradle.process.internal.ExecHandleRunner.run(ExecHandleRunner.java:68)
	... 2 more
Caused by: java.io.IOException: Cannot run program "git" (in directory "C:\Users\Ryan\Documents\GitHub\Robotics\GRIP"): CreateProcess error=2, The system cannot find the file specified
	at net.rubygrapefruit.platform.internal.DefaultProcessLauncher.start(DefaultProcessLauncher.java:25)
	... 5 more
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	... 6 more
[sts] Time taken: 0 min, 3 sec
[sts] -----------------------------------------------------```
Previous master (201dcbb6dafc8eff79464b9102069125ba5d076f) works.

Wil update once I can run `git bisect`.

rlee287, 2016-01-31:
Update:
21a3fd144652969efce745673f0a5aea148f470d is the first bad commit
commit 21a3fd144652969efce745673f0a5aea148f470d
Author: Jonathan Leitschuh <Jlleitschuh@wpi.edu>
Date:   Fri Jan 22 17:13:19 2016 -0500

    Add AutoVersioning to gradle

    Closes #414

:100644 100644 9b05e523081ca81366f16beebc5da2030f526e2a ee7e440c1caab26bb42b740aa8a1502a5e66e728 M      build.gradle

ThomasJClark, 2016-01-31:
Yeah, I was pretty sure that commit would break the build. @JLLeitschuh insisted otherwise. It looks like we're trying to run git to get the current version of GRIP, and it's not being done in a way that always works on Windows.

JLLeitschuh, 2016-02-01:
Its because you didn't install git bash on your path. When you install gitbash you need to make sure that you check the box that adds git to your path.

IntelliJ won't let you open the project either if you don't have git on your path.

rlee287, 2016-02-01:
Is there any way to prevent the code from automatically getting the latest version? (Sometimes I like to go back to different commits of GRIP to test)

ThomasJClark, 2016-02-01:
Sorry, by "get the current version", I meant "compute a string representing the version you currently have checked out".

Usually, projects have something like this in their build system that lets them create installers and stuff with a number appended to the end (like grip-1.1.1.exe or whatever):

>     version = "1.1.1"

Instead, so we don't have to update that line for every release, we're basically doing this:

>     version = $(git describe --tags)

This only works if `git` is in the path, so it's been working on our machines and the CI servers, but it doesn't work for the general case.

@JLLeitschuh can you fix this in the build?

JLLeitschuh, 2016-02-01:
The source code has to be within a git repository directory.
Bouncing between commits is fine. When you went through the git installer you need to have checked the box that adds git to your path, or you can add it to your path automatically.

This is caused by the same sort of thing as not having your `JAVA_HOME` set correctly.

ThomasJClark, 2016-02-01:
That means that the ZIPs/tarballs that GitHub generates for each release won't work, though.

JLLeitschuh, 2016-02-01:
So we can add a gradle argument that lets you manually specify the version number you want to use I guess.

@JacisNonsense is this something you want to do?

JacisNonsense, 2016-02-01:
I'll see what I can do

rlee287, 2016-02-01:
For teams that would prefer to simply use Gradle's integration with Eclipse, how would they specify Gradle arguments from the Gradle Build dialog, which looks like this:
[image]
Would it just go under the "arguements" tab?

JacisNonsense, 2016-02-01:
@rlee287 I don't have much experience with Eclipse, but I would assume as such. I'll give it a test once I implement the feature.

@JLLeitschuh Can you assign this to me?

JLLeitschuh, 2016-02-01:
Thanks @JacisNonsense you are awesome!

rlee287, 2016-02-06:
Even though I added git to my path, I am still getting errors when building the latest version (11388ac).
A dialog box comes up saying
```CreateProcess error=2, The system cannot find the file specified
Could not fetch model of type 'BuildInvocations' using Gradle distribution 'https://services.gradle.org/distributions/gradle-2.10-bin.zip'.```
Full stack trace:
```
:buildSrc:compileJava UP-TO-DATE
:buildSrc:extractSources UP-TO-DATE
:buildSrc:compileGroovy UP-TO-DATE
:buildSrc:processResources UP-TO-DATE
:buildSrc:classes UP-TO-DATE
:buildSrc:jar UP-TO-DATE
:buildSrc:assemble UP-TO-DATE
:buildSrc:compileTestJava UP-TO-DATE
:buildSrc:compileTestGroovy UP-TO-DATE
:buildSrc:processTestResources UP-TO-DATE
:buildSrc:testClasses UP-TO-DATE
:buildSrc:test UP-TO-DATE
:buildSrc:check UP-TO-DATE
:buildSrc:build UP-TO-DATE

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred evaluating root project 'GRIP'.
> A problem occurred starting process 'command 'git''

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 1.948 secs
```

PR #488 (2016-03-14): Enable non-git builds
Upon merging this will resolve #461 

For users without Git on their PATH, build versions will default to the current HEAD Commit Tag read manually. Alternatively, users may also specify the 'vers' property (via `./gradlew task -Pvers=myversion`) if they want to tag the build with a specified version.

For users with Git on their PATH, the behavior of grabbing the current Tag remains unchanged.
closed 2016-03-14 by JLLeitschuh