WPIRoboticsProjects/GRIP #462: APPCRASH on Windows10

MichaelMai2000, 2016-02-01:
Version=1
EventType=APPCRASH
EventTime=130985397406412163
ReportType=2
Consent=1
UploadTime=130985397408543888
ReportIdentifier=a0cc918e-c679-11e5-82a1-7845c4b20401
IntegratorReportIdentifier=8fc577e1-8a0d-48f6-9523-0c3cdafc2f08
NsAppName=GRIP.exe
Response.BucketId=ce7d8c416be8355f0bbfbdf7702c611e
Response.BucketTable=4
Response.LegacyBucketId=120402111127
Response.type=4
Sig[0].Name=应用程序名
Sig[0].Value=GRIP.exe
Sig[1].Name=应用程序版本
Sig[1].Value=0.0.0.0
Sig[2].Name=应用程序时间戳
Sig[2].Value=56143266
Sig[3].Name=故障模块名称
Sig[3].Value=packager.dll
Sig[4].Name=故障模块版本
Sig[4].Value=0.0.0.0
Sig[5].Name=故障模块时间戳
Sig[5].Value=5614326c
Sig[6].Name=异常代码
Sig[6].Value=c0000005
Sig[7].Name=异常偏移
Sig[7].Value=000000000001d6f0
DynamicSig[1].Name=OS 版本
DynamicSig[1].Value=10.0.10586.2.0.0.256.48
DynamicSig[2].Name=区域设置 ID
DynamicSig[2].Value=2052
DynamicSig[22].Name=其他信息 1
DynamicSig[22].Value=0bdd
DynamicSig[23].Name=其他信息 2
DynamicSig[23].Value=0bdd44dad4d107e7a7658363130eec93
DynamicSig[24].Name=其他信息 3
DynamicSig[24].Value=43fe
DynamicSig[25].Name=其他信息 4
DynamicSig[25].Value=43fe827c1aeae216da1b7a2568ae60db
UI[2]=C:\Users\天承\AppData\Local\GRIP\GRIP.exe
UI[3]=GRIP.exe 已停止工作
UI[4]=Windows 可以联机检查该问题的解决方案。
UI[5]=联机检查解决方案并关闭该程序
UI[6]=以后联机检查解决方案并关闭程序
UI[7]=关闭程序
LoadedModule[0]=C:\Users\天承\AppData\Local\GRIP\GRIP.exe
LoadedModule[1]=C:\WINDOWS\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\WINDOWS\system32\KERNEL32.DLL
LoadedModule[3]=C:\WINDOWS\system32\KERNELBASE.dll
LoadedModule[4]=C:\WINDOWS\system32\USER32.dll
LoadedModule[5]=C:\WINDOWS\system32\GDI32.dll
LoadedModule[6]=C:\WINDOWS\system32\SHELL32.dll
LoadedModule[7]=C:\WINDOWS\system32\msvcrt.dll
LoadedModule[8]=C:\WINDOWS\system32\cfgmgr32.dll
LoadedModule[9]=C:\WINDOWS\system32\windows.storage.dll
LoadedModule[10]=C:\WINDOWS\system32\combase.dll
LoadedModule[11]=C:\WINDOWS\system32\RPCRT4.dll
LoadedModule[12]=C:\WINDOWS\system32\bcryptPrimitives.dll
LoadedModule[13]=C:\WINDOWS\system32\advapi32.dll
LoadedModule[14]=C:\WINDOWS\system32\sechost.dll
LoadedModule[15]=C:\WINDOWS\system32\shlwapi.dll
LoadedModule[16]=C:\WINDOWS\system32\kernel.appcore.dll
LoadedModule[17]=C:\WINDOWS\system32\shcore.dll
LoadedModule[18]=C:\WINDOWS\system32\powrprof.dll
LoadedModule[19]=C:\WINDOWS\system32\profapi.dll
LoadedModule[20]=C:\Users\天承\AppData\Local\GRIP\MSVCP120.dll
LoadedModule[21]=C:\Users\天承\AppData\Local\GRIP\MSVCR120.dll
LoadedModule[22]=C:\WINDOWS\system32\IMM32.DLL
LoadedModule[23]=C:\Users\天承\AppData\Local\GRIP\packager.dll
LoadedModule[24]=C:\Users\天承\AppData\Local\GRIP\runtime\bin\server\jvm.dll
LoadedModule[25]=C:\WINDOWS\system32\PSAPI.DLL
LoadedModule[26]=C:\WINDOWS\SYSTEM32\WINMM.dll
LoadedModule[27]=C:\WINDOWS\SYSTEM32\WSOCK32.dll
LoadedModule[28]=C:\Users\天承\AppData\Local\GRIP\MSVCR100.dll
LoadedModule[29]=C:\WINDOWS\system32\WS2_32.dll
LoadedModule[30]=C:\WINDOWS\SYSTEM32\WINMMBASE.dll
LoadedModule[31]=C:\WINDOWS\SYSTEM32\VERSION.dll
State[0].Key=Transport.DoneStage1
State[0].Value=1
FriendlyEventName=已停止工作
ConsentKey=APPCRASH
AppName=GRIP.exe
AppPath=C:\Users\天承\AppData\Local\GRIP\GRIP.exe
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=D8EDFA81851EA13090A3E22DD5F711A4

Report.wer is as above.
What is the problem?

JLLeitschuh, 2016-02-01:
Can you paste your GRIP.log file?
Thanks!

MichaelMai2000, 2016-02-01:
@JLLeitschuh Thank you for reply, but where can I find GRIP.log?

JLLeitschuh, 2016-02-01:
In your home directory.

For Microsoft Windows Vista, 7, 8 and 10 it would be in `<root>\Users\<username>`

MichaelMai2000, 2016-02-02:
I opened my home directory, but I didn't found it.
I searched "GRIP" in my root disk, but I still couldn't find any file named as GRIP.log
[image]

MichaelMai2000, 2016-02-13:
Can anybody help me?...
@JLLeitschuh

JLLeitschuh, 2016-02-15:
The file should be in in something like C://users/[your user name]/GRIP.log

MichaelMai2000, 2016-02-16:
[image]
My user name is "天承"(2 Chinese characters), I can't find GRIP.log in C://users/天承 .
I think it is because my user name is Chinese and GRIP haven't got time to create GRIP.log but suddenly crash.
Anybody have met this situation before?
@JLLeitschuh

JLLeitschuh, 2016-02-16:
@ThomasJClark Could this be related to character encoding?

JLLeitschuh, 2016-02-16:
Can you run this command into your command line and send us the output?
`chcp`

MichaelMai2000, 2016-02-17:
I run "chcp" in cmd.exe.
Here is the output:
Microsoft Windows [版本 10.0.10586]
(c) 2015 Microsoft Corporation。保留所有权利。

C:\Users\天承>chcp
活动代码页: 936

C:\Users\天承>

JLLeitschuh, 2016-02-17:
For reference: 
https://msdn.microsoft.com/en-us/library/windows/desktop/dd317756%28v=vs.85%29.aspx

 Identifier | .NET Name | Additional information
 --- | --- | ---
| 936 | gb2312 | ANSI/OEM Simplified Chinese (PRC, Singapore); Chinese Simplified (GB2312)

JLLeitschuh, 2016-02-17:
Possibly related: http://bugs.java.com/view_bug.do?bug_id=8146306

JLLeitschuh, 2016-02-17:
@MichaelMai2000 Does your computer have a touch screen?

Related touchscreen fix that may resolve this issue: http://stackoverflow.com/questions/31786980/javafx-windows-10-combobox-error

MichaelMai2000, 2016-02-17:
@JLLeitschuh 
No, My computer doesn't have a touch screen.

AustinShalit, 2016-07-15:
Possibly related to #594

closed 2020-09-08 by MichaelMai2000