WPIRoboticsProjects/GRIP #465: attempting to connect a blur to an RGB Threshold

xjschwen, 2016-02-02:
Please describe what actions we can take to reproduce the bug you found, including the error message.

Thread[JavaFX Application Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.IllegalArgumentException: Cannot connect sockets
	at com.google.common.base.Preconditions.checkArgument(Preconditions.java:122)
	at edu.wpi.grip.core.Connection.<init>(Connection.java:37)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
	at com.google.inject.internal.DefaultConstructionProxyFactory$1.newInstance(DefaultConstructionProxyFactory.java:86)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:105)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
	at com.google.inject.internal.InjectorImpl$2$1.call(InjectorImpl.java:1016)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
	at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1012)
	at com.google.inject.assistedinject.FactoryProvider2.invoke(FactoryProvider2.java:770)
	at com.sun.proxy.$Proxy8.create(Unknown Source)
	at edu.wpi.grip.ui.pipeline.SocketHandleView.lambda$null$79(SocketHandleView.java:128)
	at java.util.Optional.ifPresent(Optional.java:159)
	at edu.wpi.grip.ui.pipeline.SocketHandleView.lambda$new$80(SocketHandleView.java:108)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$DnDGesture.fireEvent(Scene.java:2933)
	at javafx.scene.Scene$DnDGesture.processTargetDrop(Scene.java:3159)
	at javafx.scene.Scene$DnDGesture.access$6400(Scene.java:2909)
	at javafx.scene.Scene$DropTargetListener.drop(Scene.java:2873)
	at com.sun.javafx.tk.quantum.GlassSceneDnDEventHandler.lambda$handleDragDrop$310(GlassSceneDnDEventHandler.java:95)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassSceneDnDEventHandler.handleDragDrop(GlassSceneDnDEventHandler.java:92)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleDragDrop$364(GlassViewEventHandler.java:672)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleDragDrop(GlassViewEventHandler.java:671)
	at com.sun.glass.ui.View.handleDragDrop(View.java:712)
	at com.sun.glass.ui.View.notifyDragDrop(View.java:1037)
	at com.sun.glass.ui.gtk.GtkDnDClipboard.pushToSystemImpl(Native Method)
	at com.sun.glass.ui.gtk.GtkDnDClipboard.pushToSystem(GtkDnDClipboard.java:39)
	at com.sun.glass.ui.SystemClipboard.flush(SystemClipboard.java:51)
	at com.sun.glass.ui.ClipboardAssistance.flush(ClipboardAssistance.java:59)
	at com.sun.javafx.tk.quantum.QuantumClipboard.flush(QuantumClipboard.java:274)
	at com.sun.javafx.tk.quantum.QuantumToolkit.startDrag(QuantumToolkit.java:1224)
	at javafx.scene.Scene$DnDGesture.dragDetectedProcessed(Scene.java:2953)
	at javafx.scene.Scene$DnDGesture.process(Scene.java:3022)
	at javafx.scene.Scene$DnDGesture.access$8200(Scene.java:2909)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3773)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$355(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$50(GtkApplication.java:139)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.1.1-37-g39f1158
java.version | 1.8.0_65
javafx.version | 8.0.65
os.name | Linux
os.version | 3.19.0-32-generic
os.arch | amd64

JLLeitschuh, 2016-02-02:
Can you provide more information?
 - Were you loading from a file when this happened?
 - Can you provide a screenshot of your pipeline when this happened?
 - Can you reproduce this problem reliably?
 - Do you have a save file that we can use to reproduce this problem?

xjschwen, 2016-02-03:
I am sorry to report that I am a bad user.

I was attempting to connect the blur output to the RGB input button when
this happened.

I have not been able to reproduce this error and I do not have a screen
shot.   I will work harder to provide more information next time.

Jeff

AustinShalit, 2016-07-29:
@xjschwen I am going to close this issue for now.  If you come across this error again please reopen this issue.  

In the [latest release](https://github.com/WPIRoboticsProjects/GRIP/releases) of GRIP, you can only connect sockets that are connectable!

closed 2016-07-29 by AustinShalit