WPIRoboticsProjects/GRIP #473: Running on an ARM device with GUI

Gbear605, 2016-02-05:
This issue is more for future reference than it is in hope of a solution.

I've been trying to get GRIP with run with a GUI on the Jetson TK1. It appears that it is impossible.

The GRIP GUI relies on JavaFX. However, in early 2015, with JDK 8u33, Oracle removed JavaFX from their Java package for ARM devices, which the Jetson TK1 is. Without this support, GRIP will not compile or run.

Possible solutions:
- Run an older (earlier than JDK 8u33) version of Java
- Use OpenJDK instead of Oracle Java

JLLeitschuh, 2016-02-05:
Travis doesn't support the openjdk so this wouldn't work for our build system.
https://docs.travis-ci.com/user/languages/java

You need a "new" version of the JDK for JavaFX to work correctly.

I think this is one on Oracle to fix.

rich-coe, 2016-02-15:
This posting [http://www.chiefdelphi.com/forums/showthread.php?t=143516](http://www.chiefdelphi.com/forums/showthread.php?t=143516) describes instructions for building openjfx on the jetson.

AustinShalit, 2016-07-29:
@Gbear605 We are not going to be able to fix this one.  We recommend running GRIP in headless mode on the Jetson.  I am going to close this issue for now.  If you feel like you have new information please reopen it with a comment.

closed 2016-07-29 by AustinShalit