WPIRoboticsProjects/GRIP #479: GRIP runs virtually, fails when deployed

Arhowk, 2016-02-06:
I have a GRIP file that works fine if I just run it from the computer and load data over the dashboard. If I try to deploy it from the laptop, I get this bunch of errors

[need to get message from other computer]

When I run it on the RIO, I get these errors

~~~
12:30:23.905 PM
   WARNING: The CV dilate operation did not perform correctly.  OpenCV Error: Assertion failed (m.dims >= 2) in Mat, file /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp, line 441  	at edu.wpi.grip.core.Main.main(Main.java:41)  	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)  	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)  	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)  	at 

12:30:23.926 PM
 Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:33 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:33 PM edu.wpi.grip.core.Main onExceptionEvent  Feb 06, 2016 5:28:31 PM edu.wpi.grip.core.Main start  CONFIG: Configuration done.  12:28:17:984 INFO    : nt                  : Server 0xb3f302d0 entered connection state: GOT_CONNECTION_FROM_CLIENT  12:28:17:507 INFO    : wpilib              : WPILib version 2016.0.0  Initialization Successfulrc  12:30:17:965 INFO    : nt                  : Server 0xb3f54310 entered connection state: CONNECTED_TO_CLIENT  Initialization Started  12:30:17:785 INFO    : wpilib              : HAL base version 2016.0.0; roboRIO platform version 2016.0.0  âž” Launching Â«'/usr/local/bin/python3' '-u' '-O' '/home/lvuser/py/robot.py' 'run'Â»  INFO: NetworkTables: Dispatcher.cpp:392 client: CONNECTED to server 0.0.0.0 port 1735  12:28:36:791 INFO    : nt                  : Server 0xb3f30f30 entered connection state: GOT_CONNECTION_FROM_CLIENT  12:28:36:284 INFO    : nt                  : Server 0xb3f30cd0 entered connection state: SERVER_ERROR: Client Connected with bad protocol revision: 0x300  INFO: Exception Cleared Event  INFO: Exception Cleared Event  INFO: Exception Cleared Event  12:28:35:493 INFO    : nt                  : Close: Server 0xb3f30b90  INFO: SUCCESS! The project is running in headless mode!  12:28:35:465 INFO    : nt                  : Server 0xb3f30b90 entered connection state: GOT_CONNECTION_FROM_CLIENT  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionClearedEvent  	at edu.wpi.grip.core.Main.main(Main.java:41)  	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)  	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)  	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)  	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)  	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)  	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)  	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)  	at com.google.common.eventbus.EventBus.post(EventBus.java:275)  	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)  	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)  	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)  	at edu.wpi.grip.core.Step.onInputSocketChanged(Step.java:156)  	at edu.wpi.grip.core.Operation.perform(Operation.java:61)  	at org.bytedeco.javacpp.opencv_imgproc.erode(Native Method)  java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp:441: error: (-215) m.dims >= 2 in function Mat  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Main onExceptionEvent  	at edu.wpi.grip.core.Main.main(Main.java:41)  	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)  	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)  	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)  	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)  	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)  	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)  	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)  	at com.google.common.eventbus.EventBus.post(EventBus.java:275)  	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)  	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)  	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)  	at edu.wpi.grip.core.Step.onInputSocketChanged(Step.java:156)  	at edu.wpi.grip.core.Operation.perform(Operation.java:61)  	at org.bytedeco.javacpp.opencv_imgproc.erode(Native Method)  java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp:441: error: (-215) m.dims >= 2 in function Mat  Feb 06, 2016 5:28:34 PM edu.wpi.grip.core.Step runPerformIfPossible  INFO: Exception Cleared Event    	at edu.wpi.grip.core.Main.start(Main.java:59)  	at edu.wpi.grip.core.serialization.Project.open(Project.java:68)  	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1190)  	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)  	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)  	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)  	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)  	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)  	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)  	at edu.wpi.grip.core.serialization.ConnectionConverter.unmarshal(ConnectionConverter.java:58)  	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)  	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)  	at java.lang.reflect.Method.invoke(Method.java:483)  	at sun.reflect.GeneratedMethodAccessor2.invoke(Unknown Source)  	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:137)  	at edu.wpi.grip.generated.opencv_imgproc.Dilate.perform(Dilate.java:63)    SEVERE: The CV dilate operation did not perform correctly.    	at edu.wpi.grip.core.Main.start(Main.java:59)  	at edu.wpi.grip.core.serialization.Project.open(Project.java:68)  	at com.thoughtwork [... cut ...]
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="2" socket="0" previewed="true"/>
      <grip:Input step="3" socket="0"/>
    </grip:Connection>
  </connections>
  <settings>
    <teamNumber>1684</teamNumber>
    <publishAddress></publishAddress>
    <deployAddress>roborio-1684-frc.local</deployAddress>
    <deployDir>/home/lvuser</deployDir>
    <deployUser>lvuser</deployUser>
    <deployJavaHome>/usr/local/frc/JRE/</deployJavaHome>
  </settings>
</grip:Pipeline>
~~~

rlee287, 2016-02-07:
Try running GRIP from a console or somewhere where you will be able to capture GRIP's output (I personally use Eclipse to build from source and watch the console log) on a laptop. Whenever GRIP initially loads a file or deploys, it prints errors like what you described (I believe the devs are currently working to fix this).

I would verify that [NetworkTables are being published properly once deployed](http://wpilib.screenstepslive.com/s/4485/m/50711/l/479908-reading-array-values-published-by-networktables). If it is, then the errors can be ignored.

JLLeitschuh, 2016-02-07:
Try the newest release.
https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.2.0

Arhowk, 2016-02-08:
1.2.0 gives me this

~~~
/usr/local/frc/JRE//bin/java null -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'
Error: Could not find or load main class null

~~~

JLLeitschuh, 2016-02-08:
Can you open a new issue about that. I'll label that high priority.
Also include your save file with it.

Arhowk, 2016-02-09:
Still nothing after using the workaround. This is the last error (if it is
any different... doesnt seem to be). NetConsole is totally empty and the
process that I have running the GRIP project is stopped.

~~~
7:00:17.292 PM
 INFO: Pipeline Starting  Feb 09, 2016 12:00:17 AM
edu.wpi.grip.core.PipelineRunner$1 startUp  INFO: SUCCESS! The project is
running in headless mode!
7:00:17.392 PM
 Feb 09, 2016 12:00:17 AM edu.wpi.grip.core.Step runPerformIfPossible
7:00:17.428 PM
    at java.lang.Thread.run(Thread.java:745)   at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
  at
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
  at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)   at
java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)   at
com.google.common.util.concurrent.Callables$3.run(Callables.java:100)   at
com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
  at
edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)   at
edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)   at
edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)   at
edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)   at
edu.wpi.grip.core.operations.composite.HSVThresholdOperation.perform(HSVThresholdOperation.java:71)
 java.lang.IllegalArgumentException: HSV Threshold needs a 3-channel input
 WARNING: The HSV Threshold operation did not perform correctly.
7:00:17.452 PM
    at java.lang.Thread.run(Thread.java:745)   at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
  at
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
  at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)   at
java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)   at
com.google.common.util.concurrent.Callables$3.run(Callables.java:100)   at
com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
  at
edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)   at
edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)   at
edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)   at
edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)   at
edu.wpi.grip.core.operations.composite.HSVThresholdOperation.perform(HSVThresholdOperation.java:71)
 java.lang.IllegalArgumentException: HSV Threshold needs a 3-channel input
 SEVERE: The HSV Threshold operation did not perform correctly.  Feb 09,
2016 12:00:17 AM edu.wpi.grip.core.Main onExceptionEvent
7:00:17.607 PM
 OpenCV Error: Assertion failed (m.dims >= 2) in Mat, file
/home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp,
line 441
7:00:17.626 PM
    at java.lang.Thread.run(Thread.java:745)   at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
  at
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
  at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)   at
java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)   at
com.google.common.util.concurrent.Callables$3.run(Callables.java:100)   at
[... 225 lines cut ...]
  at
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
  at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)   at
java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)   at
com.google.common.util.concurrent.Callables$3.run(Callables.java:100)   at
com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
  at
edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)   at
edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)   at
edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)   at
edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)   at
edu.wpi.grip.core.Operation.perform(Operation.java:61)   at
edu.wpi.grip.generated.opencv_imgproc.Dilate.perform(Dilate.java:63)   at
org.bytedeco.javacpp.opencv_imgproc.dilate(Native Method)
 java.lang.RuntimeException:
/home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp:441:
error: (-215) m.dims >= 2 in function Mat  WARNING: The CV dilate operation
did not perform correctly.  Feb 09, 2016 12:00:18 AM edu.wpi.grip.core.Step
runPerformIfPossible  OpenCV Error: Assertion failed (m.dims >= 2) in Mat,
file
/home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/core/src/matrix.cpp,
line 441     at java.lang.Thread.run(Thread.java:745)   at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)

~~~

Arhowk, 2016-02-10:
Here's a version of the console hooked up to the riolog... (if its worth anything, the process never finishes and unplugging the camera posts no messages from the process)

http://pastebin.com/CvXmxZRv

sorry sick of github

closed 2017-01-08 by SamCarlberg