WPIRoboticsProjects/GRIP #485: Cannot use "Find Contours" as input to "CV Rectangle"?

gartaud, 2016-02-07:
Hi
Working on the FRC 2016 challenge, we are trying to highlight the targets found by GRIP in our smart dashboard.
One would think that a combination of "Find Contours", "CV Rectangle" and "Publish Video" would do the trick but it does not for a few reasons:
1. It appears that there is no operation that can take contours as an input except "Publish Contours"
2. "CV Rectangle" requires a scalar as input for the color... and it seems that there is no operation that outputs a scalar acceptable as input to "CV Rectangle".
At the minimum could you make the color a user-defined value so that one can draw static rectangles on images before publishing them to the smart dashboard?
Thanks!

ThomasJClark, 2016-02-07:
This is probably a good use of the [GRIP SmartDashboard plugin](https://github.com/WPIRoboticsProjects/GRIP-SmartDashboard/releases)

As a note, "Convex Hulls" and "Filter Contours" also take contours as an input.

gartaud, 2016-02-07:
Thank you! It seems that the GRIP SmartDashboard Extension should work for us.

However, I still think that GRIP itself should support doing what the extensions do...

And regardless, how does one specify the color parameter for CV Rectangle? What step in GRIP produces a scalar value that is suitable as input to CV Rectangle?

Thank you in advance for your reply

gartaud, 2016-02-07:
@ThomasJClark any idea about how to specify the color for the 
"CV Rectangle" operation? Thanks!

ThomasJClark, 2016-02-07:
I don't think you can.  CV Rectangle is pretty useless, but @JLLeitschuh wanted to include it.

gartaud, 2016-02-07:
Hi @JLLeitschuh
If one cannot specify the color that is a mandatory parameter to "CV Rectangle", then the step is indeed totally useless but I suppose that either I am missing something... or that the color parameter is supposed to be a user-defined parameter (and that somehow it does not show as so because of a bug in GRIP) isn't it?

JLLeitschuh, 2016-02-07:
We originally wanted to be able to specify color.
#51
However, its broken on Linux.
Perhaps its been fixed in the newer release of javafx

gartaud, 2016-02-08:
A color picker would be great, but if not possible simply specifying the color as r,g,b would do too imho. Thanks.

JLLeitschuh, 2016-02-08:
I just reopened #51
It was very, very out of date. That should fix this problem.

gartaud, 2016-02-08:
Great. So we should get CV Rectangle usable in the next release of GRIP?
In a matter of weeks?
Thanks!

JLLeitschuh, 2016-02-08:
What is the use case that you hope to achieve using the color picker?

gartaud, 2016-02-08:
Initially we wanted to draw contours. But because it seems we won't be able to pass the contours to cv rectangle we just want to be able to draw lines using rectangles with one size being zero. We don't really care too much about being able to pick a particular color, but cv rectangle requires that a color be passed as input...
Thanks!

npetrangelo, 2016-05-16:
While you work on the color picker, it might be better to assume a default color for the time being. That way we can at least use the rectangle without the color picker. I'm just assuming that's an easy task. I tried to do the same and also noticed there was no way to output contour data. The closest is printContoursReport, but it doesn't have output nodes to plug into the Rectangle.

JLLeitschuh, 2016-05-16:
I have a solution for this that is being worked on with @SamCarlberg

gartaud, 2017-03-30:
Hi @JLLeitschuh @SamCarlberg 
Was the solution ever implemented?
Thanks!