WPIRoboticsProjects/GRIP #489: CV resize operation did not perform correctly

billkearson, 2016-02-07:
Running Grip 1.2.0 and it works as expected on driver station, no issues.

When we deploy to the RobotRio we get the following output with error:
CV resize operation did not perform  #correctly.

Here is the output from the Deploy window:

/usr/local/frc/JRE//bin/java -Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'
Loading Dependency Injection Framework
Feb 07, 2016 8:53:20 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Feb 07, 2016 8:53:21 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1-39-g9fff323
platform: /Linux/arm/
Feb 07, 2016 8:53:25 PM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: src must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Input must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: Value must have a value to run this step.
OpenCV Error: Assertion failed (ssize.area() > 0) in resize, file /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp, line 3208
Feb 07, 2016 8:53:27 PM edu.wpi.grip.core.Step runPerformIfPossible
WARNING: The CV resize operation did not perform correctly.
java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp:3208: error: (-215) ssize.area() > 0 in function resize

	at org.bytedeco.javacpp.opencv_imgproc.resize(Native Method)
	at edu.wpi.grip.generated.opencv_imgproc.Resize.perform(Resize.java:58)
	at edu.wpi.grip.core.Operation.perform(Operation.java:61)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:137)
	at edu.wpi.grip.core.Step.onInputSocketChanged(Step.java:156)
	at sun.reflect.GeneratedMethodAccessor2.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:483)
	at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:95)
	at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:154)
	at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:80)
	at com.google.common.util.concurrent.MoreExecutors$DirectExecutor.execute(MoreExecutors.java:456)
	at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:76)
	at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:119)
	at com.google.common.eventbus.EventBus.post(EventBus.java:215)
	at edu.wpi.grip.core.serialization.ConnectionConverter.unmarshal(ConnectionConverter.java:58)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1206)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1190)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1061)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:68)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:62)
	at edu.wpi.grip.core.Main.start(Main.java:59)
	at edu.wpi.grip.core.Main.main(Main.java:41)

Feb 07, 2016 8:53:28 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: The CV resize operation did not perform correctly.
java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp:3208: error: (-215) ssize.area() > 0 in function resize

ThomasJClark, 2016-02-07:
Try using the "Resize Image" operation instead of the low level CV one.

billkearson, 2016-02-07:
Thanks for the quick response. I changed to teh "Resize Image" operation and now this is the deploy output:
/usr/local/frc/JRE//bin/java -Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'
Loading Dependency Injection Framework
Feb 07, 2016 10:13:28 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Feb 07, 2016 10:13:28 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.2.0
platform: /Linux/arm/
Feb 07, 2016 10:13:33 PM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
Feb 07, 2016 10:13:35 PM edu.wpi.grip.core.Main start
INFO: SUCCESS! The project is running in headless mode!
Feb 07, 2016 10:13:35 PM edu.wpi.grip.core.PipelineRunner$1 startUp
INFO: Pipeline Starting
OpenCV Error: Assertion failed (ssize.area() > 0) in resize, file /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp, line 3208
Feb 07, 2016 10:13:36 PM edu.wpi.grip.core.Step runPerformIfPossible
WARNING: The Resize Image operation did not perform correctly.
java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp:3208: error: (-215) ssize.area() > 0 in function resize

	at org.bytedeco.javacpp.opencv_imgproc.resize(Native Method)
	at edu.wpi.grip.core.operations.composite.ResizeOperation.perform(ResizeOperation.java:88)
	at edu.wpi.grip.core.Operation.perform(Operation.java:61)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)
	at edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Feb 07, 2016 10:13:36 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: The Resize Image operation did not perform correctly.
java.lang.RuntimeException: /home/javacpp-presets/opencv/cppbuild/linux-frc/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp:3208: error: (-215) ssize.area() > 0 in function resize

	at org.bytedeco.javacpp.opencv_imgproc.resize(Native Method)
	at edu.wpi.grip.core.operations.composite.ResizeOperation.perform(ResizeOperation.java:88)
	at edu.wpi.grip.core.Operation.perform(Operation.java:61)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)
	at edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Feb 07, 2016 10:13:36 PM edu.wpi.grip.core.Step runPerformIfPossible
WARNING: The HSV Threshold operation did not perform correctly.
java.lang.IllegalArgumentException: HSV Threshold needs a 3-channel input
	at edu.wpi.grip.core.operations.composite.HSVThresholdOperation.perform(HSVThresholdOperation.java:71)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)
	at edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
[... 255 lines cut ...]
	at java.lang.Thread.run(Thread.java:745)

Feb 07, 2016 10:13:36 PM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: The CV dilate operation did not perform correctly.
[... cut ...]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Feb 07, 2016 10:13:36 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$16
INFO: NetworkTables: Dispatcher.cpp:392 client: CONNECTED to server 10.0.86.96 port 1735

ThomasJClark, 2016-02-07:
How are you capturing images?  It seems like you're getting an empty frame

billkearson, 2016-02-08:
From the Axis M1011 Ethernet camera. If the frame was empty or null wouldn't it just be ignored and wait for the next good frame?

AdamuKaapan, 2016-02-09:
I am also getting this problem with a same error (if my memory serves me correctly from my experiments yesterday).

Arhowk, 2016-02-10:
We are having the same issue, axis m1013 but we aren't doing any resizing. Seems like GRIP can't interact with an axis IP camera at all.

JLLeitschuh, 2016-02-12:
These are the same issue I think #494