WPIRoboticsProjects/GRIP #491: Could not find main class "null"

Arhowk, 2016-02-08:
As of 1.2.0, I get this console output if I try to deploy GRIP to the roboRIO

~~~
/usr/local/frc/JRE//bin/java null -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'
Error: Could not find or load main class null
~~~

GRIP file

~~~
<grip:Pipeline>
  <sources>
    <grip:Camera>
      <property name="address" value="http://axis-camera.local/mjpg/video.mjpg"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="HSV Threshold">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1">
        <value>
          <double>63.129496402877706</double>
          <double>101.21212121212123</double>
        </value>
      </grip:Input>
      <grip:Input step="0" socket="2">
        <value>
          <double>174.28057553956833</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Input step="0" socket="3">
        <value>
          <double>123.83093525179854</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="CV erode">
      <grip:Input step="1" socket="0"/>
      <grip:Input step="1" socket="1"/>
      <grip:Input step="1" socket="2"/>
      <grip:Input step="1" socket="3">
        <value>1.0</value>
      </grip:Input>
      <grip:Input step="1" socket="4">
        <value>BORDER_CONSTANT</value>
      </grip:Input>
      <grip:Input step="1" socket="5"/>
      <grip:Output step="1" socket="0" previewed="false"/>
    </grip:Step>
    <grip:Step name="CV dilate">
      <grip:Input step="2" socket="0"/>
      <grip:Input step="2" socket="1"/>
      <grip:Input step="2" socket="2"/>
      <grip:Input step="2" socket="3">
        <value>17.0</value>
      </grip:Input>
      <grip:Input step="2" socket="4">
        <value>BORDER_CONSTANT</value>
      </grip:Input>
      <grip:Input step="2" socket="5"/>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Find Contours">
      <grip:Input step="3" socket="0"/>
      <grip:Input step="3" socket="1">
        <value>false</value>
      </grip:Input>
      <grip:Output step="3" socket="0" previewed="false"/>
    </grip:Step>
    <grip:Step name="Filter Contours">
      <grip:Input step="4" socket="0"/>
      <grip:Input step="4" socket="1">
        <value>400.0</value>
      </grip:Input>
      <grip:Input step="4" socket="2">
        <value>75.0</value>
      </grip:Input>
      <grip:Input step="4" socket="3">
        <value>0.0</value>
      </grip:Input>
      <grip:Input step="4" socket="4">
        <value>1000.0</value>
      </grip:Input>
      <grip:Input step="4" socket="5">
        <value>0.0</value>
      </grip:Input>
      <grip:Input step="4" socket="6">
        <value>1000.0</value>
      </grip:Input>
      <grip:Input step="4" socket="7">
        <value>
          <int>0</int>
          <int>100</int>
        </value>
      </grip:Input>
      <grip:Output step="4" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Publish ContoursReport">
      <grip:Input step="5" socket="0"/>
      <grip:Input step="5" socket="1">
        <value>stupidGarbage</value>
      </grip:Input>
      <grip:Input step="5" socket="2">
        <value>true</value>
      </grip:Input>
      <grip:Input step="5" socket="3">
        <value>true</value>
      </grip:Input>
      <grip:Input step="5" socket="4">
        <value>true</value>
      </grip:Input>
      <grip:Input step="5" socket="5">
        <value>true</value>
      </grip:Input>
      <grip:Input step="5" socket="6">
        <value>true</value>
      </grip:Input>
      <grip:Input step="5" socket="7">
        <value>true</value>
      </grip:Input>
    </grip:Step>
  </steps>
  <connections>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="2" socket="0" previewed="true"/>
      <grip:Input step="3" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="3" socket="0" previewed="false"/>
      <grip:Input step="4" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="4" socket="0" previewed="true"/>
      <grip:Input step="5" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output source="0" socket="0" previewed="true"/>
      <grip:Input step="0" socket="0"/>
    </grip:Connection>
  </connections>
  <settings>
    <teamNumber>1684</teamNumber>
    <publishAddress>roborio-1684-frc.local</publishAddress>
    <deployAddress>roborio-1684-frc.local</deployAddress>
    <deployDir>/home/lvuser</deployDir>
    <deployUser>lvuser</deployUser>
    <deployJavaHome>/usr/local/frc/JRE/</deployJavaHome>
  </settings>
</grip:Pipeline>
~~~

ThomasJClark, 2016-02-08:
As a workaround, go to Tools > Settings and set "Deploy JVM Options" to `-Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError`

Arhowk, 2016-02-09:
Worked, for now. Back to old issue.

PR #495 (2016-02-12): Fix deserializing ProjectSettings from previous GRIP versions
When opening projects with missing settings, we should use the default
values for those settings.  By default, XStream sets them to null, but
we can override that using `PureJavaReflectionProvider`.

Closes #491
commit d279ed47 (2016-02-12): Fix deserializing ProjectSettings from previous GRIP versions

When opening projects with missing settings, we should use the default
values for those settings.  By default, XStream sets them to null, but
we can override that using `PureJavaReflectionProvider`.

Closes #491
closed 2016-02-12 by JLLeitschuh