WPIRoboticsProjects/GRIP #492: How to auto-load our .grip file

skypanther, 2016-02-09:
We're attempting to run grip on our driver's station. We have the basics working, but we have to open the .grip file from the menus every time we start grip. I cannot find documentation on how to automatically load the file and our attempts to guess at command-line parameters that might do so have not worked.

ThomasJClark, 2016-02-09:
There are no command line parameters to open a file with the GUI currently.

You could run it in headless mode by running `java -jar core-1.2.0-all.jar [file.grip]`, or you could also use the keyboard shortcut (Ctrl+O) for convenience.  An optional command line parameter sounds like a good idea though for the next release.

skypanther, 2016-02-09:
Thanks. We'll try the `java -jar` technique. Since you've tagged this as an enhancement, I'll leave this issue open. Feel free to close if you'd rather track the idea in a different way.

PR #498 (2016-02-15): Add command line parsing in GUI mode and headless Windows support
In GUI mode, a file name can be specified to open, and --headless can be
added to the command to make it not really open the GUI.

In headless mode, we don't really need to listen for hangup signals
(since the wrapper script will kill any old instances of GRIP), so we
can remove that to make it work on non-unix systems.

Closes #493
Closes #492
closed 2016-02-15 by JLLeitschuh