WPIRoboticsProjects/GRIP #494: HSV threshold needs 3 channel input after installing 1.2.0

SupremeSteak1, 2016-02-11:
We get the following error when trying to deploy after installing version 1.2.0:
`/usr/local/frc/JRE//bin/java -Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'
Loading Dependency Injection Framework
Jan 02, 1970 2:01:53 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 02, 1970 2:01:54 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.2.0
platform: /Linux/arm/
Jan 02, 1970 2:01:58 AM edu.wpi.grip.core.Main start
INFO: Loading file /home/lvuser/project.grip
Jan 02, 1970 2:02:00 AM edu.wpi.grip.core.Main start
INFO: SUCCESS! The project is running in headless mode!
Jan 02, 1970 2:02:00 AM edu.wpi.grip.core.PipelineRunner$1 startUp
INFO: Pipeline Starting
Jan 02, 1970 2:02:00 AM edu.wpi.grip.core.Step runPerformIfPossible
WARNING: The HSV Threshold operation did not perform correctly.
java.lang.IllegalArgumentException: HSV Threshold needs a 3-channel input
	at edu.wpi.grip.core.operations.composite.HSVThresholdOperation.perform(HSVThresholdOperation.java:71)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)
	at edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Jan 02, 1970 2:02:00 AM edu.wpi.grip.core.Main onExceptionEvent
SEVERE: The HSV Threshold operation did not perform correctly.
java.lang.IllegalArgumentException: HSV Threshold needs a 3-channel input
	at edu.wpi.grip.core.operations.composite.HSVThresholdOperation.perform(HSVThresholdOperation.java:71)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:139)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:196)
	at edu.wpi.grip.core.PipelineRunner.access$200(PipelineRunner.java:32)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:76)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Jan 02, 1970 2:02:01 AM edu.wpi.grip.core.Main onExceptionClearedEvent
INFO: Exception Cleared Event
Jan 02, 1970 2:02:01 AM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$16
INFO: NetworkTables: Dispatcher.cpp:392 client: CONNECTED to server 127.0.0.1 port 1735
`

ThomasJClark, 2016-02-11:
Can you post a screenshot of your pipeline?

SupremeSteak1, 2016-02-11:
[image]
I will not be at robotics again until 5:00 on Friday, so I can't get any specific information, but this is what I have at home.

ThomasJClark, 2016-02-12:
The problem seems to be that when there's an issue connecting to the camera, the pipeline goes ahead and runs anyways, and the steps run into errors because they keep getting empty images.

CraigJSmith, 2016-02-14:
I have the exact same problem. In my testing , it occurs while running on the roboRIO or on a Raspberry Pi. I've tried it with USB webcams, axis cameras, various MJPG streaming software; but nothing seems to fix it. It must be more than just running with an invalid camera, because I've tried so many cameras with so many different configurations. It works completely fine using GRIP in the GUI mode, so it seems like the problem is exclusive to headless mode.

tomatac, 2016-02-24:
Is there a way to debug this issue?

Arhowk, 2016-02-25:
Is there a fix for this upcoming? 1.3.0rc1 did not resolve the issue.

tomatac, 2016-02-25:
Did you try to deploy 1.3.0rc1 on the RoboRio or on the Rasberry Pi? 
Do we know a possible root cause? Is there a way to debug this issue?

Arhowk, 2016-02-25:
RoboRio. The only information I can gather is that it is due to the Axis IP Camera

JLLeitschuh, 2016-02-26:
Can you attach a "Get Mat Info" operation to the output of your axis camera input and tell me what the different values display.
[image]

Unfortunately, the frame rate is broken in 1.3.0-rc1 (my fault, I fixed it already) or I'd have you post that as well.

tomatac, 2016-02-26:
We use a Microsoft Lifecam HD-3000 and have the same issue

Arhowk, 2016-02-26:
I won't have access to the robot until next week but will this be visible in the console output? The operation runs perfectly fine on the desktop

JLLeitschuh, 2016-02-26:
You'll have to publish the values to network tables.