WPIRoboticsProjects/GRIP #500: Is it possible to run GRIP on a roboRIO using a USB camera connected to the roboRIO?

jkoehring, 2016-02-13:
I have built a GRIP project on my laptop using a Microsoft LifeCam HD-3000 connected directly to my laptop. I would now like to deploy that project to a roboRIO and run it with the LifeCam connected directly to the roboRIO. Is that possible? If so, what determines the Webcam id that should be used for GRIP to access the camera on the roboRIO? Thank you.

ThomasJClark, 2016-02-13:
Yes. The webcam id will probably be 0 if you only have one webcam connected.

fireXtract, 2016-02-22:
I can't get this to work, when I look in the web UI for the roboRIO it shows up as cam1 but when I deploy to the rio it says no camera is at that slot, but when i use cam0 it says camera server started but the entire pipeline fails.

JLLeitschuh, 2016-02-23:
Does it run out of memory?

fireXtract, 2016-02-23:
Upon further testing, I found out that after a robot reboot, the cam turns on and through the USB webcam viewer I can watch a laggy version of the USB cam feed, but my grip smartdash extension doesn't use the publish video feed or the contours. Also outline viewer shows no values. I am going to stick with axis Camera over network and do local processing on PC for now, and hope VLAN helps fix the issues I had at my local week0 competition. (I saw other teams axis cams on my dash instead of mine, and I couldn't point to the axis Camera over the fms-lite formatted bridge.)

closed 2017-01-08 by SamCarlberg