WPIRoboticsProjects/GRIP #501: Feature Request: Be able to start and stop pipeline from within program

rlee287, 2016-02-14:
Being able to start and stop the pipeline from inside an FRC program would help the program save resources and run GRIP only when actually needed.

JLLeitschuh, 2016-02-15:
This is not a planned feature. You could stop and start the running grip process using your FRC user program.

ThomasJClark, 2016-02-15:
> This is not a planned feature. You could stop and start the running grip process using your FRC user program.

GRIP takes a couple seconds to start up though, and we already have starting and stopping implemented.  We could very easily make it so you can, say, set a networktables key like "GRIP/run" to true or false to start and stop the pipeline

PR #505 (2016-02-16): Start and stop headless GRIP using NetworkTables
This lets robot programs start and stop GRIP to save CPU without having
to wait for the process to start up again.

For example,

```java
NetworkTable table = NetworkTable.getTable("GRIP");

table.putBoolean("run", false); // Stop GRIP without killing the process

doSomeStuffThatDoesntNeedVision();

table.putBoolean("run", true); // Start GRIP again
```

This should be useful for teams that want to do auto alignment in teleop, since they could start and stop vision processing whenever the driver presses the button to auto aim.

Closes #501
closed 2016-02-16 by JLLeitschuh