WPIRoboticsProjects/GRIP #503: Cannot open IP Cameras in GRIP

cyberx696, 2016-02-14:
After building the latest source code, I still encounter the issue where the IP camera throws an exception and (in my case) don't even start. I've tried them in the v1.1.1 release too, but the same error occurs.

GRIP.log.txt

TECHNICAL DETAILS:
OS: Windows 7
Release: Stable exe_x64
Camera: Axis M1013, Panasonic VP104
Setup: Axis camera connected via ethernet to RoboRIO. Laptop running GRIP connected over Wifi to RoboRIO.

The first one (Axis M1013) crashed with a camera source error where it failed to grab the image. I tested this URL (http://10.6.96.10/cgi-bin/mjpeg) in a browser but it failed in GRIP. The error is below.

Feb 06, 2016 3:59:52 PM edu.wpi.grip.core.sources.CameraSource$1 run
WARNING: Failed to grab image
org.bytedeco.javacv.FrameGrabber$Exception: The camera stream ended unexpectedly
at edu.wpi.grip.core.sources.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:137)
at edu.wpi.grip.core.sources.CameraSource$1.run(CameraSource.java:227)
at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:60)
at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.EOFException: The camera stream ended unexpectedly
at edu.wpi.grip.core.sources.IPCameraFrameGrabber.readImage(IPCameraFrameGrabber.java:177)
at edu.wpi.grip.core.sources.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:130)
... 4 more
The second camera I tested (a Panasonic V104) also failed to start. I tested two different URLs (http://10.6.96.21/axis-cgi/mjpg/video.cgi?camera=1&resolution=320x240, http://10.6.96.21/mjpg/1/video.mjpg) but both returned the following stack trace:

org.bytedeco.javacv.FrameGrabber$Exception: Server returned HTTP response code: 401 for URL: http://10.6.96.21/axis-cgi/mjpg/video.cgi?camera=1&resolution=320x240
at edu.wpi.grip.core.sources.IPCameraFrameGrabber.start(IPCameraFrameGrabber.java:99)
at edu.wpi.grip.core.sources.CameraSource$1.startUp(CameraSource.java:210)
at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:54)
at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: Server returned HTTP response code: 401 for URL: http://10.6.96.21/axis-cgi/mjpg/video.cgi?camera=1&resolution=320x240
at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1840)
at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1441)
at edu.wpi.grip.core.sources.IPCameraFrameGrabber.start(IPCameraFrameGrabber.java:96)
... 4 more
I'm not sure why both times GRIP failed to load the video stream as they work perfectly fine in a browser. Some things I've tried include disabling my firewall/antivirus and disabling logins on both the cameras.

Any help would be appreciated.

**UPDATE**
Still not working. Tried putting admin username and password in the URL as so: http://admin:12345@10.6.96.10/cgi-bin/mjpeg.  

It still fails to even open the stream with the new radio. Please let me know if you have any idea what the issue what might be.

Here's the latest stack trace, with GRIPv1.2:

org.bytedeco.javacv.FrameGrabber$Exception: The camera stream ended unexpectedly
    at edu.wpi.grip.core.sources.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:137)
    at edu.wpi.grip.core.sources.CameraSource$1.run(CameraSource.java:227)
    at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:60)
    at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
    at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.EOFException: The camera stream ended unexpectedly
    at edu.wpi.grip.core.sources.IPCameraFrameGrabber.readImage(IPCameraFrameGrabber.java:177)
    at edu.wpi.grip.core.sources.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:130)
    ... 4 more

cyberx696, 2016-02-19:
I was wondering if anyone else is having this issue. It's towards the end of build season and I'd like to be able to use GRIP, but I need to know whether it's going to work or not very soon. 

Thanks

JLLeitschuh, 2016-02-19:
For the axis camera try just using the address: `http://10.6.96.10` and see if that works.

JLLeitschuh, 2016-02-19:
If that doesn't work then there is a release candidate that may work.
https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.3.0-rc1

cyberx696, 2016-02-20:
@JLLeitschuh 

I got the Axis working, thanks. But do you know why I can't get the Panasonic VP104 cam from working? I can view it without any authentication at http://10.6.96.10/cgi-bin/mjpeg. But when I paste that exact URL into GRIP, it crashes. I also tried just the IP in GRIP, but that doesn't work either. With the Panasonic, it requires the install of some viewer software before use.  I suspect that it may be causing issues with GRIP, but I can't be sure. 

Any help would be appreciated.

JLLeitschuh, 2016-02-21:
It seems that the panasonic cameras need to have each frame requested. Unlike cameras that simply supporting mjpg streams.
Does this camera work with smart dashboard?

cyberx696, 2016-02-26:
@JLLeitschuh 

No, when I add the IP in the SmartDashboard, it does not work. It does have a mjpg stream that I can view at IP/cgi-bin/mjpeg, but I cannot open it up that way in GRIP. Given that it works in a web browser, is there anything that I can do to open it in GRIP? (By the way, I am able to get it working with a generic OpenCV video program).

JLLeitschuh, 2016-02-27:
The best thing I can tell you to do is download the GRIP source code and try to figure out why your IP address isn't working. Use the debugger, see what the http message returned actually looks like and see if IP frame grabber is parsing it wrong.

Sorry, thats probably not the answer you want to hear.

AustinShalit, 2016-07-29:
I am going to close this issue because it has grown stale.  Please reopen this issue if you still have questions or have new information to provide.

closed 2016-07-29 by AustinShalit