WPIRoboticsProjects/GRIP #506: OutOfMemory Errors

robert1356, 2016-02-15:
Ever since upgrading to 1.2, I am have extreme trouble getting GRIP to run stable.  I has some problems under 1.1.1.  Essentially, I am getting out of memory or our of space errors.  The out of space errors generally come because the out of memory error causes either GRIP or robot code to core dump and the core dump take up all the disk space on the roborio.

My process in the robot code:
RobotCode runs
I check a radio button in the chooser
if it is set to RUN - launch GRIP from the robot code
if it is set to STOP - run the gripkill script (1st two lines from your run grip script from v1.2)

I have attached:
the riolog trace
the hs_err file specified by the OOM exception
the grip project we're using

[GRIP OOM erro - riolog.txt](https://github.com/WPIRoboticsProjects/GRIP/files/131485/GRIP.OOM.erro.-.riolog.txt)

[replay_pid10034.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/131486/replay_pid10034.log.txt)
[findTargets.grip.txt](https://github.com/WPIRoboticsProjects/GRIP/files/131487/findTargets.grip.txt)

ThomasJClark, 2016-02-15:
Set your JVM Arguments to `-Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError`.

robert1356, 2016-02-15:
That is already being done.  It's in the launch script.  See the riolog.txt file

robert1356, 2016-02-15:
Another example with a simpler project and this includes hs_err and replay_pid

[hs_err_pid3230.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/131545/hs_err_pid3230.log.txt)
[replay_pid3230.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/131546/replay_pid3230.log.txt)

robert1356, 2016-02-15:
[publishResizedVideo.grip.txt](https://github.com/WPIRoboticsProjects/GRIP/files/131554/publishResizedVideo.grip.txt)

Sn3akyP3t3, 2016-02-18:
Adobe ColdFusion uses the JVM and has excellent documentation over the years on JVM tuning and also identifying memory leaks.  While the tutorials are geared for the ColdFusion Markup Language, CFML, its bound to have some helpful tips to aid the FRC developers and GRIP maintainers to identify problems and tuning their JVM for their needs or setting a baseline.  

http://www.adobe.com/devnet/coldfusion/articles/coldfusion_performance.html

Be careful when changing the Garbage Collector, GC, settings.  The GC process is a stop the world process meaning that during the GC run it no longer responds to external requests until it completes.  See some basic information and deep dive as well at https://plumbr.eu/handbook/what-is-garbage-collection

closed 2017-01-08 by SamCarlberg