WPIRoboticsProjects/GRIP #516: Stereo Vison

madson812, 2016-02-20:
Please implement stereo vision from Open CV.

JLLeitschuh, 2016-02-21:
I'm kind of curious about this. I'll give it a shot. Any particular methods?
I'm looking into this one right now http://docs.opencv.org/master/dd/d53/tutorial_py_depthmap.html

JLLeitschuh, 2016-02-21:
Were you thinking something like this?
[image]

madson812, 2016-02-24:
That's exactly what we want! Would you mind pushing that to github?

ThomasJClark, 2016-02-24:
This won't work unless we also have the alignment stuff from opencv

madson812, 2016-02-24:
Ok, thank you for your time!

JLLeitschuh, 2016-02-24:
Do you have a stereo camera or were you just planning on putting two cameras next to each other?

warrenlp, 2016-03-21:
Using a camera like the ZED 3D camera would be extremely useful.

JLLeitschuh, 2016-03-23:
Does this work with that kind of camera?
https://github.com/WPIRoboticsProjects/GRIP/pull/517

warrenlp, 2016-03-24:
The ZED 3D camera has it's own specialized APIs to work with, but it may be possible to use stereovision from OpenCV with this camera as well.