WPIRoboticsProjects/GRIP #521: centerX value not being published

SupremeSteak1, 2016-02-23:
We have GRIP mostly working, but while trying to get x centering working on our aiming system, we realized that the centerX value is being pushed as a single number, 405, not an array of updating numbers as it should be. We have checked all our code and there does not seem to be an issue there.

ThomasJClark, 2016-02-23:
Could you verify this using outline viewer?

SupremeSteak1, 2016-02-23:
Yes. We checked and the same is true in the outline viewer.

ThomasJClark, 2016-02-23:
Interesting. Can you upload screenshots of both your GRIP pipeline and OutlineViewer? That would make it a little easier to figure out the problem. 

Also, make sure you have the latest release of GRIP.

SupremeSteak1, 2016-02-23:
[image]
[image]
We are using the latest version

SupremeSteak1, 2016-02-23:
Things just got weirder...
After many failed attempts we got back to other testing and as one improvement we added code to start grip when the robot starts. When the robot was restarted, everything is working perfectly! centerX is now an array and is reported properly. This is all good, except now we cannot connect to the robot VIA the drivers station by wireless or teather. This might not be a GRIP issue, but it seems likely.

bradamiller, 2016-02-23:
Can you connect to the robot using other means, like ssh or pings or the SmartDashboard?

Also, just curious, what language is the robot program written in?

Thanks
Brad

SupremeSteak1, 2016-02-23:
We have confirmed that it is a problem with GRIP as when the code was commented out there was no connection issue, but when it is there we cannot connect. All other wireless functions work flawlessly. I can SSH in.

SupremeSteak1, 2016-02-23:
Furthermore, the robot is programmed in java.

JLLeitschuh, 2016-07-08:
Has this been resolved in a more recent version of GRIP?

SupremeSteak1, 2016-07-09:
I do not recall that we were having any problems with it recently, so I believe so, but I do not have access to the robot right now so I cannot verify. However, I may not be able to verify for several weeks because we are doing summer cleaning right now so I would assume that it is solved for now, and if I find that it is still an issue, I will let you know.

closed 2016-07-13 by JLLeitschuh