WPIRoboticsProjects/GRIP #524: GRIP Says all cameras are in use

KevinMackenzie, 2016-02-24:
My team has two cameras on our robot.  I am just using the "simple vision" example code for one camera (`"cam1"`) and I want to use GRIP for `"cam0"`.  No matter which ID I set the USB camera to GRIP, it always launches and says the camera is already in use.  I know both cameras works, because I can change the simple vision to read from `"cam0"` and it works just the same as with `"cam1"`.  I originally suspected it was doing some sort of caching, or wasn't overwriting the file when deploying, but even after deleting both `grip.java` and `project.grip` from the roboRIO, it gave the same results.

JLLeitschuh, 2016-02-24:
Have you tried restarting your RIO? Perhaps some old process has the camera open still.
What camera types are you using? Logitech?

KevinMackenzie, 2016-02-24:
yea, i've tried restarting the rio.  The camera I am using as `cam1` is the microsoft one from andymark, and the one I am trying to use as `cam0` in GRIP is a logitech c310.

JLLeitschuh, 2016-02-25:
Are you running with the newest version of grip? [1.3.0-rc1](https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.3.0-rc1)

KevinMackenzie, 2016-02-25:
yes.  Initially I tried using 1.2.0, but whenever I tried to deploy, it stopped and said the `main` function could not be found, so I went to 1.3.0-rc1.  On a slightly separate note, when I do not have the camera server running in the FRC program on `cam1`, GRIP fails and says "HSV Threshold needs a 3-channel input".

JLLeitschuh, 2016-02-25:
Oh, so the camera server is probably preventing GRIP from reading from the USB camera.
Do you get the error once or does it spam the console multiple times? If you see "Exception Cleared" after that error you should be all good and GRIP should be running successfully.

KevinMackenzie, 2016-02-25:
I can't' remember; next time I have access to the hardware I will check

AustinShalit, 2016-04-23:
@KevinMackenzie Has this issue been solved? If so, please close this issue. If not, please let us know how we can help.

KevinMackenzie, 2016-04-23:
yea, sorry for leaving this open; we ended up switched to openCV in python on a raspberry pi

closed 2016-04-23 by KevinMackenzie